adafruit/Adafruit_TinyUSB_Arduino #144: ESP32-S2 MSC Not Working Properly after WiFi Connect

brentru, 2021-12-16:
### Operating System

MacOS

### IDE version

1.8.13

### Board

Adafruit Feather ESP32-S2

### BSP version

Latest

### TinyUSB Library version

Latest

### Sketch

```
/*********************************************************************
 Adafruit invests time and resources providing this open source code,
 please support Adafruit and open-source hardware by purchasing
 products from Adafruit!

 MIT license, check LICENSE for more information
 Copyright (c) 2019 Ha Thach for Adafruit Industries
 All text above, and the splash screen below must be included in
 any redistribution
*********************************************************************/

/* This example demo how to expose on-board external Flash as USB Mass Storage.
 * Following library is required
 *   - Adafruit_SPIFlash https://github.com/adafruit/Adafruit_SPIFlash
 *   - SdFat https://github.com/adafruit/SdFat
 *
 * Note: Adafruit fork of SdFat enabled ENABLE_EXTENDED_TRANSFER_CLASS and FAT12_SUPPORT
 * in SdFatConfig.h, which is needed to run SdFat on external flash. You can use original
 * SdFat library and manually change those macros
 */

#include "SPI.h"
#include "SdFat.h"
#include "Adafruit_SPIFlash.h"
#include "Adafruit_TinyUSB.h"
#include <WiFi.h>

// Uncomment to run example with FRAM
// #define FRAM_CS   A5
// #define FRAM_SPI  SPI

#if defined(FRAM_CS) && defined(FRAM_SPI)
  Adafruit_FlashTransport_SPI flashTransport(FRAM_CS, FRAM_SPI);

#elif defined(ARDUINO_ARCH_ESP32)
  // ESP32 use same flash device that store code.
  // Therefore there is no need to specify the SPI and SS
  Adafruit_FlashTransport_ESP32 flashTransport;

#else
  // On-board external flash (QSPI or SPI) macros should already
  // defined in your board variant if supported
  // - EXTERNAL_FLASH_USE_QSPI
  // - EXTERNAL_FLASH_USE_CS/EXTERNAL_FLASH_USE_SPI
  #if defined(EXTERNAL_FLASH_USE_QSPI)
    Adafruit_FlashTransport_QSPI flashTransport;

  #elif defined(EXTERNAL_FLASH_USE_SPI)
    Adafruit_FlashTransport_SPI flashTransport(EXTERNAL_FLASH_USE_CS, EXTERNAL_FLASH_USE_SPI);

  #else
    #error No QSPI/SPI flash are defined on your board variant.h !
  #endif
#endif

Adafruit_SPIFlash flash(&flashTransport);

// file system object from SdFat
FatFileSystem fatfs;

FatFile root;
FatFile file;
File myFile;

// USB Mass Storage object
Adafruit_USBD_MSC usb_msc;

// Set to true when PC write to flash
bool fs_changed;

// Network
char ssid[] = "YOUR_SSID_HERE";   // your network SSID
char key[] = "YOUR_SSID_KEY_HERE"; // your network key
int status = WL_IDLE_STATUS;

// the setup function runs once when you press reset or power the board
void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);

  flash.begin();

  // Set disk vendor id, product id and revision with string up to 8, 16, 4 characters respectively
  usb_msc.setID("Adafruit", "External Flash", "1.0");

  // Set callback
  usb_msc.setReadWriteCallback(msc_read_cb, msc_write_cb, msc_flush_cb);

  // Set disk size, block size should be 512 regardless of spi flash page size
  usb_msc.setCapacity(flash.size()/512, 512);

  // MSC is ready for read/write
  usb_msc.setUnitReady(true);
  
  usb_msc.begin();

  // Init file system on the flash
  fatfs.begin(&flash);

  Serial.begin(115200);
 // while ( !Serial ) delay(10);   // wait for native usb

  Serial.println("Adafruit TinyUSB Mass Storage External Flash example");
  Serial.print("JEDEC ID: 0x"); Serial.println(flash.getJEDECID(), HEX);
  Serial.print("Flash size: "); Serial.print(flash.size() / 1024); Serial.println(" KB");

  // open the file
  myFile = fatfs.open("test.txt", FILE_WRITE);

  // if the file opened okay, write to it:
  if (myFile) {
    Serial.print("Writing to test.txt...");
    myFile.println("testing, pre-connect");
    myFile.flush();
    delay(50);
    // close the file:
    myFile.close();
    Serial.println("done.");
  } else {
    // if the file didn't open, print an error:
    Serial.println("error opening test.txt");
  }

  // Connect to Network
  Serial.println("Connecting, calling WiFi.begin()");
  // attempt to connect to Wifi network:
  while ( WiFi.status() != WL_CONNECTED) {
    Serial.println("Attempting to connect to network");
    status = WiFi.begin(ssid, key);
    delay(5000);
  }
  Serial.println("Connected!");

  // open the file
  myFile = fatfs.open("test.txt", FILE_WRITE);

  // if the file opened okay, write to it:
  if (myFile) {
    Serial.print("Writing to test.txt...");
    myFile.println("testing, post-connect");
    // close the file:
    myFile.flush();
    delay(50);
    myFile.close();
    Serial.println("done.");
  } else {
    // if the file didn't open, print an error:
    Serial.println("error opening test.txt");
  }

  // WiFi Disconnect
  WiFi.disconnect(true);

  // open the file
  myFile = fatfs.open("test.txt", FILE_WRITE);

  // if the file opened okay, write to it:
  if (myFile) {
    Serial.print("Writing to test.txt...");
    myFile.println("testing post-disconnect");
    // close the file:
    myFile.close();
    Serial.println("done.");
  } else {
    // if the file didn't open, print an error:
    Serial.println("error opening test.txt");
  }

  
}

void loop()
{
// no-op
}

// Callback invoked when received READ10 command.
// Copy disk's data to buffer (up to bufsize) and 
// return number of copied bytes (must be multiple of block size) 
int32_t msc_read_cb (uint32_t lba, void* buffer, uint32_t bufsize)
{
  // Note: SPIFLash Bock API: readBlocks/writeBlocks/syncBlocks
  // already include 4K sector caching internally. We don't need to cache it, yahhhh!!
  return flash.readBlocks(lba, (uint8_t*) buffer, bufsize/512) ? bufsize : -1;
}

// Callback invoked when received WRITE10 command.
// Process data in buffer to disk's storage and 
// return number of written bytes (must be multiple of block size)
int32_t msc_write_cb (uint32_t lba, uint8_t* buffer, uint32_t bufsize)
{
  digitalWrite(LED_BUILTIN, HIGH);

  // Note: SPIFLash Bock API: readBlocks/writeBlocks/syncBlocks
  // already include 4K sector caching internally. We don't need to cache it, yahhhh!!
  return flash.writeBlocks(lba, buffer, bufsize/512) ? bufsize : -1;
}

// Callback invoked when WRITE10 command is completed (status received and accepted by host).
// used to flush any pending cache.
void msc_flush_cb (void)
{
  // sync with flash
  flash.syncBlocks();

  // clear file system's cache to force refresh
  fatfs.cacheClear();

  fs_changed = true;

  digitalWrite(LED_BUILTIN, LOW);
}
```

### What happened ?

After the WiFi radio has been initialized via a `WiFi.begin(ssid, key);` call, writing to the filesystem no longer works. Opening the `test.txt` file on the example sketch's filesystem only shows "testing, pre-connect". Occasionally, rebooting the device entirely will show the "post-connect/disconnect" lines in the test.txt file.

Calling `WiFi.disconnect()` does not seem to resolve the issue.

### How to reproduce ?

1. Edit the modified `msc_external_flash` example
2. Run the modified `msc_external_flash` example
3. Check the filesystem's `test.txt` file

brentru, 2021-12-16:
Hacking on this more.. resetting the device will reproducibly show an updated test.txt file with contents from after the begin() call.

hathach, 2021-12-20:
@brentru will test this out real soon, is porting tinyuf2 for S3 :)

hathach, 2022-01-04:
@brentru I have just tested this, this is actually not a bug of TinyUSB lib. The file is already written to the spi flash, however host driver (Windows/Linux/macOS) always assume it is the only one making changes to the filesystem and have an comprehensive caching with block device. 

  spi flash <-----> TinyUSB <-----> host caching <-----> host driver

When making change to spi flash internally, we must also request host driver to flush/update its cache. Unfortunately, I also don't know any way to do so (probably need more effort on researching) except for detach (disconnect) and re-attach (re-connect). See also 
- https://github.com/hathach/tinyusb/discussions/584

Hope this explanation is clear enough to you, let me know if there is anything you want to clear i up.

**Note**: when firmware is writing flash device and enumerated as msc device, there is a potential conflict, host driver such as macos can do logging/indexing in the background which could cause flash corruption in the FAT table. CircuitPython resolve this by having an mode switch that only allow 1 side writing to the flash (other will be read-only). Here is the sum-up diagram

closed 2022-01-10 by hathach

hathach, 2022-01-17:
Update: I think test unit ready can be used as walkaround to force host to update. You could simulate and ejected event by return false to test unit ready and set sense key to NOT READY (subkey not found). Then report as ready again, it would force host to refresh. Example to use unit ready is implemented in #152

FedericoBusero, 2022-12-11:
An example of MSC with wifi can be found here 

https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/Adafruit_ESP32S2_TFT_WebServer/Adafruit_ESP32S2_TFT_WebServer.ino

I don't know if it helps