adafruit/Adafruit_TinyUSB_Arduino #145: [Feature] Add internal flash storage to MSD

ladyada, 2022-01-05:
for @brentru 's work we would like to eventually add support for exposing internal flash FS of 48KB as MSD so that boards without QSPI can store wifi connection data
we could use this library:
https://github.com/cmaglie/FlashStorage
Circuitpython uses FAT FS of 48KB so can test against SAMD21 board with circuitpy previously installed.
we only need it to with SAMD21 for now

hathach, 2022-01-06:
48KB is quite tight for FATFS, I try to find best way to implement this.

ladyada, 2022-01-06:
it is! but we have used it in circuitpython for gemma, trinket, feather m0 basic, and other older SAMD21 boards without SPI Flash so you could install circuitpython on that older boards, and then try to read the FS.

hathach, 2022-01-06:
got it

hathach, 2022-01-11:
update: pulling things together, got into an weird issue that filesystem got erased every upload. It turns out that bossac 1.7 (default upload tool) does a full chip erase when upload a new sketch, I am switching **ProxLight Trinkey** (as well as other trinkey) to bossac 1.8 which only erase needed sectors/pages. Seem good so far, will do more tests and we could migrate to 1.8 over time.

```
./bossac --help
Usage: bossac [OPTION...] [FILE]
Basic Open Source SAM-BA Application (BOSSA) Version 1.7.0
Flash programmer for Atmel SAM devices.
Copyright (c) 2011-2012 ShumaTech (http://www.shumatech.com)

Examples:
  bossac -e -w -v -b image.bin   # Erase flash, write flash with image.bin,
                                 # verify the write, and set boot from flash
  bossac -r0x10000 image.bin     # Read 64KB from flash and store in image.bin

Options:
  -e, --erase           erase the entire flash (keep the 8KB of bootloader for SAM Dxx)
```

ladyada, 2022-01-11:
oh yes! that sounds good

hathach, 2022-01-12:
this is implemented by, please check them out for the implementation details 
- #148 
- https://github.com/adafruit/ArduinoCore-samd/pull/316
- https://github.com/adafruit/Adafruit_InternalFlash

ladyada, 2022-01-12:
ok @brentru wanna try it?

hathach, 2022-01-13:
fully implemented, once SAMD BSP 1.7.7 is released https://github.com/adafruit/arduino-board-index/pull/93 . It can be tested with neo/neokey/slide/proxlight trinkey. The filesystem should be accessible and changes should persist when switching between arduino sketch <-> circuitpython.

closed 2022-01-13 by hathach