adafruit/Adafruit_TinyUSB_Arduino #150: ESP32S2 MassStorage with External Flash chip

1337ralfy69, 2022-01-13:
### Operating System

Others

### IDE version

1.8.19

### Board

ESP32S2

### BSP version

1.8.4

### TinyUSB Library version

github latest

### Sketch

examples/massstorage/msc_external_flash

### What happened ?

Hey there, im trying to build a usb mass storage device with ESP32S2 and W25Q80 flash chip connected via SPI and using Adafruit TinyUSB and Adafruit SPIFlash library. :star_struck:

So, msc_ramdisk and msc_sdfat(with sd card adpter) examples works great and shows up as mass storage device, also Marzogh`s SpiMemory library example TestFlash works with flash chip, shows flash ID e.t.c. :slightly_smiling_face:

But i cant figure out why example code msc_external_flash(TinyUSB) or flash_info(Adafruit SPIFlash) doesnt run on ESP32S2. Codes are similar, and i suspect that it does not correctly initialize SPI interface, as oscilliscope is connected to SCK and SS pins, all above written skeches shows activity on SPI bus, but when msc_external_flash or flash_info are uploaded, it shows floating pins. :face_with_raised_eyebrow:
And also these two example sketches with no alteration runs ok, but no SPI communication, when trying set "custom" SPI, ESP chip starts resetting itself.

`#include "SdFat.h"
#include "Adafruit_SPIFlash.h"

#define CUSTOM_CS   34
#define CUSTOM_SPI  SPI //some examples use FSPI insted of SPI, but it doesnt work here

Adafruit_FlashTransport_SPI flashTransport(CUSTOM_CS, CUSTOM_SPI);

Adafruit_SPIFlash flash(&flashTransport);

void setup() {
  Serial.begin(115200);
  Serial.println("Adafruit Serial Flash Info example");

  //  flash.begin(); //this one should be used, but esp starts reseting when calling, mby thats the problem?
  flashTransport.begin(); //initializes SPI but obviously flash.getJEDECID() and flash.size() will not work.
}

void loop() {
  Serial.print("JEDEC ID: 0x"); Serial.println(flash.getJEDECID(), HEX);
  Serial.print("Flash size: "); Serial.print(flash.size() / 1024); Serial.println(" KB");
  delay(100);
}`

### How to reproduce ?

Upload to ESP32S2 Board without any changes

hathach, 2022-01-18:
which specific board you are testing on, is it your custom hardware ?

1337ralfy69, 2022-01-18:
> 

Im testing on ESP32-S2-Saola-1

hathach, 2022-01-19:
This is not a bug of this repo, [Adafruit_SPIFlash](https://github.com/adafruit/Adafruit_SPIFlash) currently only supports accessing the files that is on the same chip with the executing code. We have no plan to support this, but we are happy to review and merge it if you want to make an PR (to Adafruit_SPIFlash).

closed 2022-01-19 by hathach