adafruit/Adafruit_TinyUSB_Arduino #161: USB product/device descriptor not working

steeley, 2022-03-21:
### Operating System

MacOS

### IDE version

arduino2

### Board

raspberry pico

### BSP version

no such menu

### TinyUSB Library version

release 1.10

### Sketch

#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

// USB MIDI object
Adafruit_USBD_MIDI usb_midi;

// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

void setup() 
{ 
  Serial.begin(115200);
    // these two string must be exactly 32 chars long
  //                                   01234567890123456789012345678912
  USBDevice.setManufacturerDescriptor("John Park GlobalMega Corp       ");
  USBDevice.setProductDescriptor     ("MIDIKiller so many spaces       ");
  // Initialize MIDI, and listen to all MIDI channels
  // This will also call usb_midi's begin()
  MIDI.begin(MIDI_CHANNEL_OMNI);
  // Attach the handleNoteOn function to the MIDI Library. It will
  // be called whenever the device receives MIDI Note On messages.
  MIDI.setHandleNoteOn(handleNoteOn);
  MIDI.setHandleNoteOff(handleNoteOff);
  MIDI.setHandleControlChange(handleCtrl);
  // wait until device mounted??
  while (!USBDevice.mounted()) delay(5);

  delay(250);
}

void loop() {
    MIDI.read(); 
}

//------------------FUNC----------------//
void handleNoteOn(byte channel, byte pitch, byte velocity) {
  // Log when a note is pressed.
  Serial.printf("Note on: channel = %d, pitch = %d, velocity - %d", channel, pitch, velocity);
  Serial.println();
}

void handleNoteOff(byte channel, byte pitch, byte velocity) {
  // Log when a note is released.
  Serial.printf("Note off: channel = %d, pitch = %d, velocity - %d", channel, pitch, velocity);
  Serial.println();
}

void handleCtrl(byte channel, byte ctrl, byte value) {
  // Serial.printf("ctrl: channel = %d, ctrl = %d, value - %d", channel, ctrl, value);
  Serial.println();

}

### What happened ?

name just shows up in Mac OS MIDI setup as ' pico', USB info does show the correct name and manufacturer

### How to reproduce ?

run the code look at audioMIDI setup on Mac and then see USB names in System Report

Spegs21, 2022-03-27:
What happens if you change the VID and PID?

steeley, 2022-03-27:
I don't see how to change the VID/PID. However, if I delete the device in OSX Audio MIDI utility then replug it, it seems to 
be working now.

hathach, 2022-03-28:
could you update 1st post to
1. provide the arduino core (BSP) version
2. remove sketch code and attached it as txt file instead.
3. try to run the stock example `midi_test` to see if that works

steeley, 2022-03-28:
done. I think replugging does update the USB descriptor. It may only be an issue if you keep changing the descriptors on the same board, then sometimes the OS doesn't immediately update the USB name.

hathach, 2022-03-28:
host driver always do its own caching for fast enumeration and driver mapping usually with vid/pid/bcd combo

closed 2022-03-28 by hathach

jhsa, 2022-08-23:
`USBDevice.setManufacturerDescriptor( My_Manufacturer);`

USBDevice doesn't even compile on the midi_test example. At least not when compiling for the LOLIN S2 MINI board. I try to use "TinyUSBDevice" instead, but it does not work. There is a bug somewhere in my opinion.