adafruit/Adafruit_TinyUSB_Arduino #162: [Feature]Compatibility with official Arduino SAMD core

Spegs21, 2022-03-27:
**Is your feature request related to a problem? Please describe.**
I've implemented USB HID  support through TinyUSB for SAMD boards in [CorsairLightingProtocol](https://github.com/Legion2/CorsairLightingProtocol) and users are wanting to use it with official Arduino boards, like the Zero (Legion2/CorsairLightingProtocol#280), but are not able to since the official Arduino core doesn't support TinyUSB.

**Describe the solution you'd like**
I'd like to bring the ability to use TinyUSB to the Arduino SAMD core. I'm not asking for full support (arduino/ArduinoCore-samd#666).

**Describe alternatives you've considered**
Use the built in USB core but it adds unnecessary complexity to the project.

**Additional context**
I originally envisioned using the USE_TINYUSB define in the official Arduino core to disable the built in USB support but I think it can be made more generic using a menu option to toggle the definition of USBCON. I've created and example [here](https://github.com/Spegs21/ArduinoCore-samd/commit/e5fc960c77509fc31804dac144e99a00bb7e2581).

TinyUSB doesn't seem to care about USBCON so I used what was in place to achieve the goal of disabling the Arduino core USB. I understand that this is probably not the intended use of this define but it works well this way.

Now come changes needed to TinyUSB. I can bypass the [`TinyUSB is not selected...`](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/5c88b7d1a5f5b46a57dfbde2c1d7425974d6112f/src/Adafruit_TinyUSB.h#L31) error by adding addition conditions to the macro:

```
#if !defined(USE_TINYUSB) && ( (defined(ARDUINO_ARCH_SAMD) && defined(ARDUINO_SAMD_ADAFRUIT)) || \
                               (defined(ARDUINO_ARCH_RP2040) && !defined(ARDUINO_ARCH_MBED)) )
#error TinyUSB is not selected, please select it in "Tools->Menu->USB Stack"
#endif
```

Finally, since USE_TINYUSB is not defined the tusb_config_samd.h needs updated [here](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/5c88b7d1a5f5b46a57dfbde2c1d7425974d6112f/src/arduino/ports/samd/tusb_config_samd.h#L41):

`#if defined(USE_TINYUSB) || !defined(USBCON)`

Since the Adafruit SAMD core will still require TinyUSB to be selected and the Arduino SAMD core will not define USBCON, this should work. You could also adopt not defining USBCON in your SAMD core when TinyUSB is selected and simplify the changes made to the Arduino core implementation by only checking for USBCON and not USE_TINYUSB. USBCON now essentially means 'I want to use the Arduino USB core'.

Since I wanted to keep the changes to the Arduino SAMD core generic, this is the best why I could come up with to do this. I realize I'm not explicitly telling the library I want to use TinyUSB but I am now telling it I don't want to use the Arduino USB core. If I've gone through the process of including the library and its functions and I don't want to use the Arduino USB core, I think this should be enough for it to understand its use is desired.

I'm definitely open to other options. I want to see what you think about this @hathach.

hathach, 2022-03-28:
with above changed, have you managed to get tinyusb running on samd core ?

hathach, 2022-03-28:
build just fine is far from working. TinyUSB supported cannot be added without core modification, since core need it for USB Serial and ability to switch between arduino/tinyusb stack much like changes to our own fork. 
- wrap cores/arduino/USB with `#ifndef USE_TINYUSB`
- add menu option to switch stack
- add tinyusb as part of built-in library
- several other modification to boards.txt and platform.txt

And I don't think Arduino team would agree to adopt required changes, there is little I could do about it.  Let me know if you could persuade Arduino team to accept core change

closed 2022-03-28 by hathach

hathach, 2022-03-28:
please submit an PR for your proposal changes. I will give it a review and see if we could merge it.