adafruit/Adafruit_TinyUSB_Arduino #164: no matching function for call to 'Adafruit_USBD_HID::Adafruit_USBD_HID(const uint8_t [77], unsigned int, hid_interface_protocol_enum_t, int, bool)'

fusss9, 2022-04-06:
### Operating System

Windows 11

### IDE version

arduino 1.8.12

### Board

raspberry-pi-pico

### ArduinoCore version

1.8.12

### TinyUSB Library version

1.10.1

### Sketch (attached txt file)

Composite/mouse_ramdisk

### What happened ?

Arduino:1.8.12 (Windows 10), 开发板："Raspberry Pi Pico, 2MB (no FS), 125 MHz, Small (-Os) (standard), Disabled, Disabled, None, Adafruit TinyUSB"

F:\soft\arduino-1.8.12-windows\arduino-1.8.12\arduino-builder -dump-prefs -logger=machine -hardware F:\soft\arduino-1.8.12-windows\arduino-1.8.12\hardware -hardware C:\Users\Acer\AppData\Local\Arduino15\packages -tools F:\soft\arduino-1.8.12-windows\arduino-1.8.12\tools-builder -tools F:\soft\arduino-1.8.12-windows\arduino-1.8.12\hardware\tools\avr -tools C:\Users\Acer\AppData\Local\Arduino15\packages -built-in-libraries F:\soft\arduino-1.8.12-windows\arduino-1.8.12\libraries -libraries C:\Users\Acer\Documents\Arduino\libraries -fqbn=rp2040:rp2040:rpipico:flash=2097152_0,freq=125,opt=Small,rtti=Disabled,dbgport=Disabled,dbglvl=None,usbstack=tinyusb -vid-pid=0000_0000 -ide-version=10812 -build-path C:\Users\Acer\AppData\Local\Temp\arduino_build_135106 -warnings=all -build-cache C:\Users\Acer\AppData\Local\Temp\arduino_cache_348403 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-elf2uf2.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-elf2uf2-1.3.3-a-ed6d983.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-openocd.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-openocd-1.3.3-a-ed6d983.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-python3.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-gcc.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-gcc-1.3.3-a-ed6d983.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-pioasm.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-pioasm-1.3.3-a-ed6d983.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-mklittlefs.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-mklittlefs-1.3.3-a-ed6d983.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.3.3-a-ed6d983 -verbose C:\Users\Acer\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-1.10.0\examples\Composite\mouse_ramdisk\mouse_ramdisk.ino
F:\soft\arduino-1.8.12-windows\arduino-1.8.12\arduino-builder -compile -logger=machine -hardware F:\soft\arduino-1.8.12-windows\arduino-1.8.12\hardware -hardware C:\Users\Acer\AppData\Local\Arduino15\packages -tools F:\soft\arduino-1.8.12-windows\arduino-1.8.12\tools-builder -tools F:\soft\arduino-1.8.12-windows\arduino-1.8.12\hardware\tools\avr -tools C:\Users\Acer\AppData\Local\Arduino15\packages -built-in-libraries F:\soft\arduino-1.8.12-windows\arduino-1.8.12\libraries -libraries C:\Users\Acer\Documents\Arduino\libraries -fqbn=rp2040:rp2040:rpipico:flash=2097152_0,freq=125,opt=Small,rtti=Disabled,dbgport=Disabled,dbglvl=None,usbstack=tinyusb -vid-pid=0000_0000 -ide-version=10812 -build-path C:\Users\Acer\AppData\Local\Temp\arduino_build_135106 -warnings=all -build-cache C:\Users\Acer\AppData\Local\Temp\arduino_cache_348403 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-elf2uf2.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-elf2uf2-1.3.3-a-ed6d983.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-openocd.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-openocd-1.3.3-a-ed6d983.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-python3.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-gcc.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-gcc-1.3.3-a-ed6d983.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-pioasm.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-pioasm-1.3.3-a-ed6d983.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-mklittlefs.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.3.3-a-ed6d983 -prefs=runtime.tools.pqt-mklittlefs-1.3.3-a-ed6d983.path=C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.3.3-a-ed6d983 -verbose C:\Users\Acer\Documents\Arduino\libraries\Adafruit_TinyUSB_Arduino-1.10.0\examples\Composite\mouse_ramdisk\mouse_ramdisk.ino
Using board 'rpipico' from platform in folder: C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.13.1
Using core 'rp2040' from platform in folder: C:\Users\Acer\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.13.1
Detecting libraries used...
"C:\\Users\\Acer\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.3.3-a-ed6d983/bin/arm-none-eabi-g++" -c -Werror=return-type -DUSE_TINYUSB "-IC:\\Users\\Acer\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\1.13.1/libraries/Adafruit_TinyUSB_Arduino/src/arduino" -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x2e8a -DUSB_PID=0x000a "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions "-iprefixC:\\Users\\Acer\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\1.13.1/" "@C:\\Users\\Acer\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\1.13.1/lib/platform_inc.txt" -fno-rtti -std=gnu++17 -g -w -x c++ -E -CC -DSERIALUSB_PID=0x000a -DUSBD_MAX_POWER_MA=250 -DF_CPU=125000000L -DARDUINO=10812 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -Os "-IC:\\Users\\Acer\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\1.13.1\\cores\\rp2040" "-IC:\\Users\\Acer\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\1.13.1\\variants\\rpipico" "C:\\Users\\Acer\\AppData\\Local\\Temp\\arduino_build_135106\\sketch\\mouse_ramdisk.ino.cpp" -o nul
Alternatives for Adafruit_TinyUSB.h: [Adafruit_TinyUSB_Arduino@1.9.2 Adafruit_TinyUSB_Arduino-1.10.0@1.10.0 Adafruit_TinyUSB_Library@1.4.4]
ResolveLibrary(Adafruit_TinyUSB.h)
  -> candidates: [Adafruit_TinyUSB_Arduino@1.9.2 Adafruit_TinyUSB_Arduino-1.10.0@1.10.0 Adafruit_TinyUSB_Library@1.4.4]
"C:\\Users\\Acer\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.3.3-a-ed6d983/bin/arm-none-eabi-g++" -c -Werror=return-type -DUSE_TINYUSB "-IC:\\Users\\Acer\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\1.13.1/libraries/Adafruit_TinyUSB_Arduino/src/arduino" -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x2e8a -DUSB_PID=0x000a "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions "-iprefixC:\\Users\\Acer\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\1.13.1/" "@C:\\Users\\Acer\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\1.13.1/lib/platform_inc.txt" -fno-rtti -std=gnu++17 -g -w -x c++ -E -CC -DSERIALUSB_PID=0x000a -DUSBD_MAX_POWER_MA=250 -DF_CPU=125000000L -DARDUINO=10812 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -Os "-IC:\\Users\\Acer\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\1.13.1\\cores\\rp2040" [... cut ...]
	at processing.app.Editor.resumeOrCloseSerialMonitor(Editor.java:2107)
	at processing.app.Editor.access$1300(Editor.java:117)
	at processing.app.Editor$UploadHandler.run(Editor.java:2081)
	at java.lang.Thread.run(Thread.java:748)
Caused by: jssc.SerialPortException: Port name - COM15; Method name - openPort(); Exception type - Port not found.
	at jssc.SerialPort.openPort(SerialPort.java:167)
	at processing.app.Serial.<init>(Serial.java:136)
	... 8 more
打开串行端口COM15出错

### How to reproduce ?

I have this problem when uploading

hathach, 2022-04-06:
please update your post to have compile log as attached txt

fusss9, 2022-04-06:
> 请更新您的帖子以将编译日志作为附件 txt

I found the cause of the problem because I used a very old library version number 1.4.4
"Use version 1.4.4 of the library Adafruit_TinyUSB_Library in the folder: C:\Users\Acer\Documents\Arduino\libraries\Adafruit_TinyUSB_Library"

thank you very much!

hathach, 2022-04-06:
glad that works for you

closed 2022-04-06 by hathach