adafruit/Adafruit_TinyUSB_Arduino #168: [Feature] USB Host functionality

aschueth, 2022-05-10:
I have a USB peripheral (GPS puck) that I would like to read into an ESP32-S2 board. I naively thought this was going to be simple, but am now realizing it is quite nuanced. From what I understand, using USB_CDC is not realistic since the ESP32 board needs to be USB-host. Additionally, from what I understand, the tinyUSB library (https://github.com/hathach/tinyusb) supports USB-Host, but this library does not. I have been pointed to this library (https://github.com/chegewara/EspTinyUSB) in the meantime for USB-host functionality, but I believe it would be a good fit in this current library as well.

Chubls, 2022-06-24:
On the topic of this; 

I can see the newly added pico PIO usb host and example. I currently have a usb keyboard plugged into a pico.
I've run the example, i get the device descriptors back correctly.

How would i then go about reading keystrokes etc from the keyboard? i cant see anything that would allow this

hathach, 2022-12-05:
basic host function with hid and msc is already implemented with rp2040 using pio-usb

closed 2022-12-05 by hathach