adafruit/Adafruit_TinyUSB_Arduino #169: [Feature]Add PICO PIO USB support

sfranzyshen, 2022-05-14:
Add support for https://github.com/sekigon-gonnoc/Pico-PIO-USB

with the latest pull request to Pico-PIO-USB ... TinyUSB support has been added
https://github.com/sekigon-gonnoc/Pico-PIO-USB/pull/22 

and there is a active pull request to TinyUSB  itself ...
https://github.com/hathach/tinyusb/pull/1412

Ref.
https://github.com/earlephilhower/arduino-pico/issues/582

hathach, 2022-05-16:
we will add host support to this library, but there is always some lag between this arduino library and tinyusb upstream. Please be patient

kilograham, 2022-05-20:
please hold off on this; i would like to back-out and fix up the way the PIO support is integrated into the build. It will likely break whatever you do here.

kilograham, 2022-05-20:
I added a PR https://github.com/hathach/tinyusb/pull/1467

sfranzyshen, 2022-06-16:
looks like pull-request #173 and release 1.11.0 - Updated tinyusb to support rp2040 pio usb host handled this?

hathach, 2022-06-16:
no, it is an weekly scheduled release probably by script on all adafruit repo. and only update tinyusb core only.

closed 2022-06-20 by hathach

JonnyHaystack, 2022-06-20:
Just curious, is there something that makes implementing the host API easier with PIO USB than with normal USB?

tannewt, 2022-06-21:
> Just curious, is there something that makes implementing the host API easier with PIO USB than with normal USB?

I don't think so. The appeal of using PIO is having the USB device active at the same time.

JonnyHaystack, 2022-06-23:
> > Just curious, is there something that makes implementing the host API easier with PIO USB than with normal USB?
> 
> I don't think so. The appeal of using PIO is having the USB device active at the same time.

Hmm okay, well I hope it's planned to have host support with the built-in USB controller as well 🙏
I've been hoping to move to this library for an existing device I designed, which uses the actual USB controller for USB host, and I use the PIO for other comms so wouldn't be able to use PIO USB anyway.