adafruit/Adafruit_TinyUSB_Arduino #170: Composite HID descriptor without Serial device?

oomek, 2022-05-20:
Is it possible to remove Serial Device and Input Device from Composite HID?

[image]

Here is my  hid descriptor:

https://github.com/oomek/Pico-Dual-Gamepad/blob/master/include/hid_desc.hpp

hathach, 2022-05-20:
This feature is already implemented and possible. Serial.end() must be called before any other usb set up. For further question, please use Adafruit forum

oomek, 2022-05-20:
But the problem is this does not work as I mentioned in the other closed issue. I wouldn't have bothered you if it was that simple.
The closure of my issue was a bit premature though.

oomek, 2022-05-20:
Besides if I don't include Arduino.h i have to use `SerialTinyUSB.end()` but that does not work either.

hathach, 2022-05-21:
>  The closure of my issue was a bit premature though.

This issue is created with "Feature Request", which is already implemented and confirmed by others hence closed. If you have problems using it 
- And don't know how to use it, post a question on forum support as suggested in the template
- or believe it is a bug, then create an bug and filled all detail for reproducing.