adafruit/Adafruit_TinyUSB_Arduino #172: MultiSerial sketch does not work on SAMD51 architecture

anisoptera, 2022-05-20:
### Operating System

Linux

### IDE version

Arduino 1.8.19 / VSCode Arduino Extension

### Board

Trellis M4 Express

### ArduinoCore version

1.7.10 (Adafruit SAMD Boards)

### TinyUSB Library version

1.9.4 / 1.10.3

### Sketch (attached txt file)

examples/CDC/cdc_multi

### What happened ?

Running the basic multi serial port sketch for this library results in only one serial device being enumerated on either Windows or Linux.

I was able to get two serial ports working in CircuitPython, which doesn't use the Adafruit_TinyUSB driver.

Looking into the source code, I notice that SAMD boards are defined as having a max of `1` `CDC` device. I tried changing this `#define` to 2 (by copying the SAMD header into my project and modifying it) but it doesn't appear to change anything.

Looking at the code, I see several things that worry me, including the fact that calling `.end` on a CDC device causes the shared instance count variable to be reset to 0.

Was this tested on M4 devices?

### How to reproduce ?

1. Install sketch
2. Enumerate serial ports on connected system
3. Observe only one serial port (should be two)

hathach, 2022-06-09:
just tested with feather M4, and it works well enough, have you changed the correct config file at https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/src/arduino/ports/samd/tusb_config_samd.h

Note: if have installed the  Adafruit_TinyUSB_Arduino in the library folder, you need to update it since it will be selected instead of the cores/libraries in the BSP.

anisoptera, 2022-06-09:
What exactly did you test on a Feather? Did you have to modify any of the library files to get it to work? Because I would not consider needing to modify files (without instructions) to get the example sketch working “working well enough”.

I definitely tested on both a Trellis M4 and a PyPortal and neither one worked out of the box, nor with my attempted modifications to the most obvious file. Can you explain what you had to do to get it working?

hathach, 2022-06-09:
The sketch explicitly says you need to modify the configuration to increase the number of supported CDCs. I will add a few more text to make it clearer 

https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/CDC/cdc_multi/cdc_multi.ino#L16-L17

This feature is not often used and supporting more cdc interface requires more SRAM for tx/rx fifo and ep buffer. Therefore it is not enabled by default, unless there is enough people voting for it.

closed 2022-06-09 by hathach

anisoptera, 2022-06-10:
Fair enough, I guess I missed that text. Thanks for the pointer. :)

hathach, 2022-06-10:
no problems at all, I will make the text more obvious