adafruit/Adafruit_TinyUSB_Arduino #174: PlatformIO Compile Error: 'Adafruit_USBD_MIDI' does not name a type

SunboX, 2022-06-14:
### Operating System

MacOS

### IDE version

PlatformIO 6.0.3

### Board

az-delivery-devkit-v4

### ArduinoCore version

1.0.6

### TinyUSB Library version

1.10.3

### Sketch (attached txt file)

#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

// USB MIDI object
Adafruit_USBD_MIDI usb_midi;

// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

void setup()
{
}

void loop()
{
}

### What happened ?

Compiler stopped with Exception: 'Adafruit_USBD_MIDI' does not name a type

### How to reproduce ?

Setup PlatformIO project with the following platform.ini

```
[env:az-delivery-devkit-v4]
platform = espressif32
board = az-delivery-devkit-v4
framework = arduino
lib_deps = 
	adafruit/Adafruit TinyUSB Library@^1.10.3
	fortyseveneffects/MIDI Library@^5.0.2
monitor_speed = 115200
```

Put the Sketch into main.cpp and Build the project

### Debug Log

> Executing task: platformio run --environment az-delivery-devkit-v4 <

Processing az-delivery-devkit-v4 (platform: espressif32; board: az-delivery-devkit-v4; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/az-delivery-devkit-v4.html
[... 576 lines cut ...]
                 from .pio/libdeps/az-delivery-devkit-v4/Adafruit TinyUSB Library/src/class/vendor/vendor_device.c:27:
.pio/libdeps/az-delivery-devkit-v4/Adafruit TinyUSB Library/src/common/tusb_mcu.h:276:4: warning: #warning "TUP_DCD_ENDPOINT_MAX is not defined for this MCU, default to 8" [-Wcpp]
   #warning "TUP_DCD_ENDPOINT_MAX is not defined for this MCU, default to 8"
    ^
In file included from .pio/libdeps/az-delivery-devkit-v4/Adafruit TinyUSB Library/src/tusb_option.h:176:0,
                 from .pio/libdeps/az-delivery-devkit-v4/Adafruit TinyUSB Library/src/common/tusb_common.h:70,
                 from .pio/libdeps/az-delivery-devkit-v4/Adafruit TinyUSB Library/src/osal/osal.h:34,
                 from .pio/libdeps/az-delivery-devkit-v4/Adafruit TinyUSB Library/src/common/tusb_fifo.c:28:
.pio/libdeps/az-delivery-devkit-v4/Adafruit TinyUSB Library/src/common/tusb_mcu.h:276:4: warning: #warning "TUP_DCD_ENDPOINT_MAX is not defined for this MCU, default to 8" [-Wcpp]
   #warning "TUP_DCD_ENDPOINT_MAX is not defined for this MCU, default to 8"
    ^
==================================================================================================== [FAILED] Took 1.29 seconds ====================================================================================================
The terminal process "platformio 'run', '--environment', 'az-delivery-devkit-v4'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

### Screenshots

[image]

[image]

[image]

[image]

SunboX, 2022-06-14:
Hm, actually it seems to be included anyway. So I don't know where the issue comes from: https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/537be8c9a770f93acc90fede9b33184b3dfe76f6/src/arduino/ports/esp32/Adafruit_TinyUSB_esp32.cpp#L68

SunboX, 2022-06-14:
Ah, the code is disable by a compiler switch: https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/74383273b3e0cf5ebbd2510a2db0ea7cac3393cc/src/arduino/ports/esp32/Adafruit_TinyUSB_esp32.cpp#L45

That might explain why it's not there. So I'm lost, how should it work on an ESP32 ?

SunboX, 2022-06-14:
Maybe the MIDI stuff was disabled by accident when disabling all the general USB handler code?

hathach, 2022-06-14:
if using esp32 core, you need to use core v2. v1 is too obsolete and won't support tinyusb.

closed 2022-06-14 by hathach

ladyada, 2022-06-17:
esp32 (not s2 or s3) does not support USB native either, it will never act as a midi device