adafruit/Adafruit_TinyUSB_Arduino #176: WebUSB serial example broken (regression)

maxgerhardt, 2022-06-22:
### Operating System

Windows 10

### IDE version

1.8.16

### Board

Adafruit Feather RP2040

### ArduinoCore version

2.2.0

### TinyUSB Library version

1.12.0

### Sketch (attached txt file)

examples/WebUSB/webusb_serial/webusb_serial.ino

### What happened ?

When compiling and uploading the Web USB serial example to a RP2040 board using the [Arduino-Pico core](https://github.com/earlephilhower/arduino-pico) in the latest 2.2.0 version, which uses Adafruit TinyUSB version 1.11.0, the received text in the WebUSB Serial example page does not contain linebreaks.

When overwriting the library files in `C:\Users\<user>\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.2.0\libraries\Adafruit_TinyUSB_Arduino` with the 1.10.0 release, the problem goes away.

Thus, "something" broke in between [these two versions](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/compare/1.10.0...1.10.2).

### How to reproduce ?

1. Open the Arduino IDE
2. Install Arduino-Pico core per https://github.com/earlephilhower/arduino-pico#installing-via-arduino-boards-manager
3. Use Tools settings per Arduino IDE screenshot below
4. Open reference example sketch
5. Have a RP2040 board connected
6. Press upload
7. Sketch should upload and, on Windows, show the "Device is ready to be connected.." notification
8. Click on it to be forwarded to https://adafruit.github.io/Adafruit_TinyUSB_Arduino/examples/webusb-serial/index.html using Microsoft Edge as the browser (does **not** work in Firefox)
9. Click on "Connect"
10. Observer "Receiver" output on the right-hand window

### Debug Log

Debug log with good 1.10.0 version:
[output_good.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/8960856/output_good.txt)

Debug log with bad 1.12.0 version:
[output_bad.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/8960857/output_bad.txt)

Logs were generated by setting `CFG_TUSB_DEBUG` to `2`.

### Screenshots

Behavior in 1.10.0 (good):

[image]

Behavior in 1.10.2 up to latest 1.12.0 (bad) (confirmed in https://github.com/earlephilhower/arduino-pico/issues/642#issuecomment-1162541777):

[image]

Arduino IDE config:

[image]

hathach, 2022-06-30:
thank you very much for detail report on the issue. This is actually the landing page web issue, it didn't break line properly when multiple lines are sent in packet
- https://github.com/hathach/tinyusb-webusb-page/pull/1
it is fixed in the landing page repo, but this example used an copy version which didn't updated. Just updated the example, also add the link to landing page source to the example as well

The reason that the older version of tinyusb (hardly) does not cause this, is the new usb improve the write performance of vendor class. The later tinyusb cached and does not flush() every write (similar to Serial, it will only flush when 64 bytes are accumulated or flush() is actively called from user. Note: Serial.flush() is normally called behind the scene).

- https://github.com/hathach/tinyusb/pull/1289

I hope this clear thing up. pr #180 should fix this, if you have time to test it out, please let me know your result.

hathach, 2022-07-01:
the issue is with the sketch and the landing page. can you confirm
1. You run the updated webusb_serial in the PR
2. use the new landing page it suggest which is https://example.tinyusb.org/webusb-serial/index.html
3. If possible please test with Arduino

maxgerhardt, 2022-07-01:
Yes the PlatformIO compiled firmware works with the above `ini` when visiting https://example.tinyusb.org/webusb-serial/index.html with it -- 

[image]

however, as shown in the screenshot I **did** take the landing page [from the PR](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/fix-webusb-example/docs/examples/webusb-serial), not the adafruit website or the one from the master branch.  So what's hosted on https://example.tinyusb.org/webusb-serial/index.html is not equal to what's in the PR?

hathach, 2022-07-01:
included in docs/ is not the landing page from the sketch, it should suggest the one from tinyusb. But I will update it anyway. Thank you for confirming.

hathach, 2022-07-01:
local page is updated as well with the pr. Thank you for your issue