adafruit/Adafruit_TinyUSB_Arduino #185: Xinput support - emulate xbox 360 controllers

leecyrille, 2022-07-07:
**Is your feature request related to a problem? Please describe.**
Existing solutions for xinput and the RP2040/pico require platform.io and other unstable solutions which are difficult to use.

I am attempting to make a 4 player xinput solution for arcade cabinets

**Describe the solution you'd like**
Add xinput to the Adafruit_TinyUSB library similar to this:
[](https://github.com/fluffymadness/tinyusb-xinput)

**Describe alternatives you've considered**
I have am using David Madison's xinput library for Arduino in my arcade cabinet now, but it is slower to initilize and not as nice for flashing. RP2040 availability is also better for building custom boards.

**Additional context**

Arduino 32U4 and Teensy xinput library:
https://github.com/dmadison/ArduinoXInput
https://www.partsnotincluded.com/understanding-the-xbox-360-wired-controllers-usb-data/ (Extremely versbose explaination of xinput)
https://www.partsnotincluded.com/xbox-360-controller-led-animations-info/ (details about the player LED data - LED data is useful for trying to build multiplayer controllers)

Example project with RP2040 and xinput option (also nintendo switch):
https://github.com/FeralAI/GP2040 (older version)
https://github.com/OpenStickFoundation/GP2040-CE (Newer community fork under active development)
https://discord.gg/kbU9FC2v (Discord channel with firmware developer channel - massive amounts of info regarding game controllers)

composite libary for ST F1/F4 with xinput AND multiple Xinput on a single MCU (This would be amazing to have in arduino)
https://github.com/arpruss/USBComposite_stm32f1

ladyada, 2022-07-10:
any code submissions are welcome as examples!

leecyrille, 2022-07-12:
> any code submissions are welcome as examples!

https://github.com/FeralAI/MPG
I think this MPG library is the closest thing I can figure out right now. It can do xinput with tinyUSB on Arduino, but not with RP2040. it can do xinput on RP2040 with platform.io

RobertDaleSmith, 2022-08-05:
@leecyrille I am looking to do xinput with tinyUSB on RP2040 as well. I came across this project, but still working on verifying it works for me. [https://github.com/fluffymadness/tinyusb-xinput]( https://github.com/fluffymadness/tinyusb-xinput)

leecyrille, 2022-08-08:
> @leecyrille I am looking to do xinput with tinyUSB on RP2040 as well. I came across this project, but still working on verifying it works for me. https://github.com/fluffymadness/tinyusb-xinput

The GP2040-CE project is an advanced working example with platform.io more properly configured. You can look at my issue for the (https://github.com/fluffymadness/tinyusb-xinput) which explains how I got it working. I am mostly looking to get away from platform.io because it becomes painful if you want to edit the dependencies (which were MPG in my case).

For now I am using this:  
https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation

Here is my repo with a fully built out 2 and 4 player composite example on the bluepill, and a 4 player composite example for my custom breakoutboard with the STM32F1 chip with more pins:  
https://github.com/leecyrille/STM32_Arduino_Xinput_4Player  
https://github.com/leecyrille/STM32F103V_KiCad_4Player_Xinput_Interface

Apruss's composite library inside rogerclarkmelbourne's STM32 framework has this multi-xbox component which makes  a single hardware instance send up to six xinput reports, which makes them permenantly in the correct order inside windows somehow, which is like magic for people trying to build 4 player arcade cabinets (my project).

For the Xbox parts of that library, it seems limited to the STM32F1 chips so far (missing config somewhere for F4 chips I think).

RobertDaleSmith, 2022-08-10:
oh after digging into this more I see that this actually the opposite of what I'm looking for. Trying to find a working example of reading x-input from controllers over USB input rather than simulating x-input devices. This gets me a bit closer though. 🙏

JonnyHaystack, 2022-09-18:
I got XInput device mode working with this library and using MS OS 2.0 descriptors to get Windows to enumerate the device correctly. However, because this library defines `tud_vendor_control_xfer_cb()` in the `Adafruit_USBD_WebUSB` class, I couldn't get it working without deleting the WebUSB stuff. I certainly don't want it to be entering the WebUSB code anyway, when I'm not using that class at all. It doesn't seem possible to solve this without changes to the library itself. I would be happy to submit a PR, if a maintainer could advise the best way to proceed. @hathach ? Here's what I've got so far: https://github.com/JonnyHaystack/Adafruit_TinyUSB_XInput