adafruit/Adafruit_TinyUSB_Arduino #189: Pico W support for WIFI

fapplin, 2022-07-14:
Will this be updated to support WIFI on the Pico W?

hathach, 2022-07-14:
I don't have a picow at the moment, but do you have any problem running this library on it ?

fapplin, 2022-07-14:
I haven't tried it. The Pico W has a Infineon CYW43439 2.4 GHz Wi-Fi chip and I didn't know if it would be compatible with ESP chips. 

======================================================================= 
The price of greatness is responsibility. - Winston Churchill

fapplin, 2022-07-14:
I just got my Pico W. I plugged it in and tried to run the webusb_serial example. I got the following error. This is not a huge deal. I'm actually using the midi host aspect, but would like to have a tiny web server running for commands. This is probably not related to this project.

Arduino: 1.8.19 (Linux), Board: "Raspberry Pi Pico"

/home/frank/Arduino/libraries/Adafruit_TinyUSB_Library/examples/WebUSB/webusb_serial/webusb_serial.ino: In function 'void loop()':
webusb_serial:91:32: error: no matching function for call to 'arduino::UART::read(uint8_t [64], int)'
     count = Serial.read(buf, 64);
                                ^
In file included from /home/frank/.arduino15/packages/arduino/hardware/mbed_rp2040/3.1.1/cores/arduino/Arduino.h:102:0,
                 from sketch/webusb_serial.ino.cpp:1:
/home/frank/.arduino15/packages/arduino/hardware/mbed_rp2040/3.1.1/cores/arduino/Serial.h:51:7: note: candidate: virtual int arduino::UART::read()
   int read(void);
       ^~~~
/home/frank/.arduino15/packages/arduino/hardware/mbed_rp2040/3.1.1/cores/arduino/Serial.h:51:7: note:   candidate expects 0 arguments, 2 provided
exit status 1
no matching function for call to 'arduino::UART::read(uint8_t [64], int)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

======================================================================= 
The price of greatness is responsibility. - Winston Churchill 

 
 

Closed #189 as completed.

hathach, 2022-07-15:
please check the readme and select TinyUSB as usb stack in the IDE menu