adafruit/Adafruit_TinyUSB_Arduino #191: tuh_hid callback functions trigger compile errors

touchgadget, 2022-07-25:
### Operating System

Linux

### IDE version

Arduino 1.8.19

### Board

Raspberry Pi Pico

### ArduinoCore version

Raspberry Pi Pico/2040 by Philhower 2.3.3

### TinyUSB Library version

github 2ab122c

### Sketch (attached txt file)

[hid_test.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/9177224/hid_test.txt)

I had to rename the .ino file with .txt to make github happy.

### What happened ?

I am porting over TinyUSB examples/dual/host_hid_to_device_cdc/ to Arduino but have run into multiple definitions of tuh_hid_mount_cb.

To fix I propose removing the tuh_hid_*_cb function declarations from Adafruit_TinyUSB_Arduino.h. The functions will eventually be defined in the sketch.

The next compile error is tuh_hid_report_received_cb is undefined which is not yet defined in the sketch.

To fix this, I propose adding TU_ATTR_WEAK to tuh_hid_mount_cb and tuh_hid_report_received_cb in hid_host.h.

### How to reproduce ?

Rename hid_test.txt to hid_test.ino then compile with the IDE to see the compiler errors.

hathach, 2022-07-25:
the strong version in lib is temporary, since host support is WIP. It is addressed in #190 , I will review and revise host support when I have time.

hathach, 2022-08-01:
should be fixed now

touchgadget, 2022-08-01:
Thanks, works fine!