adafruit/Adafruit_TinyUSB_Arduino #193: FlushCDC Problem with arduino-pico core

s-t-e-p-a-n-o-v, 2022-08-09:
### Operating System

Windows 10

### IDE version

1.8.19

### Board

Raspberry PICO

### ArduinoCore version

2.3.3

### TinyUSB Library version

1.14.1

### Sketch (attached txt file)

```
#include <Adafruit_TinyUSB.h>

void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println("blink!");
  delay(200);               
}
```

### What happened ?

Even/odd loop iteration doesn't send data to PC. Data is stored in buffer and transferred on next loop cycle.

### How to reproduce ?

Compile and upload sketch with arduino-pico core with TinyUSB stack and open serial port monitor.

### Screenshots

[image]

s-t-e-p-a-n-o-v, 2022-08-09:
It's hard to find where bug is, inside arduino-pico core calls or inside TinyUSB.
Corresponding issue at arduino-pico repository
https://github.com/earlephilhower/arduino-pico/issues/720

karelv, 2022-08-19:
Windows 10 / IDE 1.8.19 / Pico 2.4.0

[image]

I observe the same behavior.

karelv, 2022-08-19:
Here is with pico sdk: (I removed the include statement)
[image]
--> Time steps of 200ms.

karelv, 2022-08-19:
In my project (fairly large code base), my observation is that the some routines make the serial interface crashing. Only some particular (application-)commands are working. Switching to pico SDK, let all work nicely. I want to use the combination of MSC and UART. This is why I want to use adafruit TinyUSB stack.   
Note: I disabled the MSC device already w/ tinyUSB, no improvement.

karelv, 2022-08-19:
I tried 1.14.1 and 1.14.3 of this lib, both have same issue.

pschatzmann, 2023-04-21:
I don't quite understand why this should be an issue. 
If you want to force the output just call Serial.flush();

A possible solution would be to add a flush(); and yield(); at the end of size_t Adafruit_USBD_CDC::write(const uint8_t *buffer, size_t size)