adafruit/Adafruit_TinyUSB_Arduino #203: Nucleo F303K8 support

SunboX, 2022-09-02:
Hi, if I try to run the TinyUSB Library against my Nucleo F303K8 board, I'm getting the following exception:

.pio/libdeps/nucleo_f303k8/Adafruit TinyUSB Library/src/tusb_config.h:46:4: error: #error TinyUSB Arduino Library does not support your core yet

The TinyUSB library itself seems to support it already:
https://github.com/hathach/tinyusb/blob/master/docs/reference/supported.rst#:~:text=dwc2-,F3,-%E2%9C%94

Is there anything I can do, to get it working?

hathach, 2022-09-08:
> Is there anything I can do, to get it working?

You literally need to do a full port for stm32
- look at how usb is implemented in the arduino stm32 core
- follow porting guide in the readme https://github.com/adafruit/Adafruit_TinyUSB_Arduino#porting-guide
- try to follow other already ports platform such as nrf,rp2040,samd etc..

hathach, 2022-12-05:
close due to lack of user response.

closed 2022-12-05 by hathach