adafruit/Adafruit_TinyUSB_Arduino #208: Boot keyboard is not able to enter BIOS

tomaszduda23, 2022-10-11:
### Operating System

Linux

### IDE version

5.2.0

### Board

lolin_s2_mini

### ArduinoCore version

2.0.5

### TinyUSB Library version

1.14.4

### Sketch (attached txt file)

[main.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/9758988/main.txt)

### What happened ?

Unable to enter BIOS without reseting board during boot-up. It can be also fixed by doing detach and attach.
```
  TinyUSBDevice.detach();
  delay(100);
  TinyUSBDevice.attach();  
```

### How to reproduce ?

1. Power on boot keyboard
2. Try to enter BIOS

### Debug Log

[boot_usb_kbd.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/9758974/boot_usb_kbd.txt)

hathach, 2023-03-22:
try to remove Serial connetion and leave only keyboard interface alone to see if that works.

hathach, 2023-11-30:
closed due to lack of user response

closed 2023-11-30 by hathach