adafruit/Adafruit_TinyUSB_Arduino #234: Support USB audio class UAC with Arduino stream

adagnano, 2023-01-05:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Bexin3, 2023-02-19:
Would be so nice to have audio examples

closed 2023-03-24 by hathach

akumpf, 2023-11-30:
@hathach Opening this back up, at least to get some clarity 😄 

Would LOVE to find a way to get USB device audio up an running (especially on ESP32-S2 and S3). I've made use of this functionality with Teensy in the past and the possibilities are endless for audio projects... both for getting lossless digital audio as an input from the computer (for effects, analysis, reactive lighting, etc.) and an output (dedicated synthesizers, instruments, etc.). 

It's also fantastic for debugging any audio project, since you can just pipe some data in/out of the USB audio callback and verify if the noise/glitch you're hearing is actually in the raw data or not.

I've found that ESP-IDF + TinyUSB has some included USB audio HOST support. But the audio DEVICE side seems to be lacking. There's also efforts like [usb_stream](https://components.espressif.com/components/espressif/usb_stream) and [ESP32_USB_STREAM](https://github.com/esp-arduino-libs/ESP32_USB_Stream), but also only addressing the HOST side.

Not expecting Adafruit to magically pull this out of a hat (🪄 🎩), but would love to make progress toward adding this functionality.

* **Any thoughts on where best to start to add USB audio DEVICE support?**
* **What existing USB device class is the closest that could be referenced/extended?** 

Thanks!