adafruit/Adafruit_TinyUSB_Arduino #235: Error compiling msc_file_explorer.ino

vanbwodonk, 2023-01-10:
### Operating System

Linux

### Arduino IDE version

Arduino 1.8.19

### Board

Raspberry Pi Pico

### ArduinoCore version

2.7.0

### TinyUSB Library version

1.17

### Sketch (attached txt file)

[examples](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/examples)/[DualRole](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/examples/DualRole)/[msc_file_explorer](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/examples/DualRole/msc_file_explorer)/msc_file_explorer.ino

### What happened ?

Error compiling msc_file_explorer.ino

### How to reproduce ?

1. Open File > Examples > Adafruit TinyUSB Library > DualRole > msc_file_explorer
2. Build

### Debug Log

```
In file included from /home/drw/.arduino15/packages/rp2040/hardware/rp2040/2.7.0/libraries/Adafruit_TinyUSB_Arduino/src/Adafruit_TinyUSB.h:77,
                 from /home/drw/.arduino15/packages/rp2040/hardware/rp2040/2.7.0/libraries/Adafruit_TinyUSB_Arduino/examples/DualRole/msc_file_explorer/msc_file_explorer.ino:31:
/home/drw/.arduino15/packages/rp2040/hardware/rp2040/2.7.0/libraries/Adafruit_TinyUSB_Arduino/src/arduino/msc/Adafruit_USBH_MSC.h:35:69: error: expected class-name before '{' token
   35 | class Adafruit_USBH_MSC_BlockDevice : public FsBlockDeviceInterface {
      |                                                                     ^
/home/drw/.arduino15/packages/rp2040/hardware/rp2040/2.7.0/libraries/Adafruit_TinyUSB_Arduino/examples/DualRole/msc_file_explorer/msc_file_explorer.ino: In function 'void tuh_msc_mount_cb(uint8_t)':
msc_file_explorer:136:28: error: cannot convert 'Adafruit_USBH_MSC_BlockDevice*' to 'BlockDevice*' {aka 'DedicatedSpiCard*'}
  136 |   is_mounted = fatfs.begin(&msc_block_dev);
      |                            ^~~~~~~~~~~~~~
      |                            |
      |                            Adafruit_USBH_MSC_BlockDevice*
In file included from /home/drw/.arduino15/packages/rp2040/hardware/rp2040/2.7.0/libraries/ESP8266SdFat/src/FatLib/FatLib.h:27,
                 from /home/drw/.arduino15/packages/rp2040/hardware/rp2040/2.7.0/libraries/ESP8266SdFat/src/SdFat.h:34,
                 from /home/drw/.arduino15/packages/rp2040/hardware/rp2040/2.7.0/libraries/Adafruit_TinyUSB_Arduino/examples/DualRole/msc_file_explorer/msc_file_explorer.ino:28:
/home/drw/.arduino15/packages/rp2040/hardware/rp2040/2.7.0/libraries/ESP8266SdFat/src/FatLib/FatVolume.h:46:27: note:   initializing argument 1 of 'bool FatVolume::begin(BlockDevice*, bool, uint8_t)'
   46 |   bool begin(BlockDevice* dev, bool setCwv = true, uint8_t part = 1) {
      |              ~~~~~~~~~~~~~^~~
/home/drw/.arduino15/packages/rp2040/hardware/rp2040/2.7.0/libraries/Adafruit_TinyUSB_Arduino/examples/DualRole/msc_file_explorer/msc_file_explorer.ino: In function 'void tuh_msc_umount_cb(uint8_t)':
msc_file_explorer:150:9: error: 'class FatVolume' has no member named 'end'
  150 |   fatfs.end();
      |         ^~~
```

hathach, 2023-01-10:
you need to install SdFat-Adafruit Fork https://github.com/adafruit/SdFat

closed 2023-01-10 by hathach

joelsernamoreno, 2023-07-15:
Hi @vanbwodonk @hathach 

I am testing this example. Is there any way to show the storage of a pendrive or anything else on computer?

I have tried the other example of reading the contents of a storage (msc_data_logger) and it works, but I have not been able to mount the storage on my laptop

Thanks

hathach, 2023-07-16:
please don't post on a closed issues especially when your problem is not EXACTLY the same as OP since this adds nothing to the issue. Open your own one, use discussion for how-to and questions, and don't tag people for your how-to question.