adafruit/Adafruit_TinyUSB_Arduino #240: msc_sdfat problem on QT py m0

akarimiz1, 2023-01-16:
### Operating System

Windows 11

### Arduino IDE version

arduino 18.13

### Board

Adafruit QT py m0

### ArduinoCore version

1.7.11

### TinyUSB Library version

1.17.0

### Sketch & Compiled output (as attached txt files)

\examples\MassStorage\msc_sdfat

Sketch uses 41884 bytes (15%) of program storage space. Maximum is 262144 bytes.

### What happened ?

Hi, I have a QT py m0 (atsamd21e18a) board. I can run msc_ramdisk tinyusb version 1.17.0 on it. It works perfectly for me. Before that, I could run msc_sdfat (16Gb) on a seeeduino xiao (atsamd21g18a) on tinyusb version 0.10.5 ( I found on the internet that tinyusb version 0.10.5 only will work on seeeduino Xiao). but the problem is that I want to run msc_sdfat on QT py m0 (atsamd21e18a), the code compiles perfectly, and I upload it on the device. But the storage is not opening and OS shows this "please insert a disk into a removable device." I checked the connection sd card reader to QT py m0 by running this example program SD -> ReadWrite, and it works. But I checked it by Sdfat-adafruit fork -> open-next it doesn't work.
Another point is I test QT py m0 by example program tinyusb msc_sd, the removable device opened, and I can open my files on it. But all options are not working. For instance, I can't copy a file into it.
All examples tinyusb msc work perfectly on seeeduino xiao (tinyusb version 0.10.5) but do not work on QT py m0 (tinyusb version 1.17.0).
last point, tinyusb version 0.10.5 doesn't compile on QT py m0.

### How to reproduce ?

1. upload msc_sdfat on QT py m0 which is connected to an SDcard reader by SPI 
2. the storage not working on windows 11 and 10

hathach, 2023-01-17:
For seeeduino xiao issue, please post it to their support forum/repo. Here we should focus on the qt py m0. Could you
1. Which SDcard reader you are using
2. Upload the picture of your hardware setup with the sdcard reader to make sure the wiring and pin config is correct.
3. did you make any modification to the msc_sdfat example, if yes then please attach the modified sketch
4. please attached the compiled output log of arduino IDE for analysis.

akarimiz1, 2023-01-17:
this is the sd card reader: https://www.icstation.com/mobile/mini-micro-card-module-memory-module-arduino-p-5043.html
and the connections are attached to the post.
exactly this is uploaded: Arduino\libraries\Adafruit_TinyUSB_Library\examples\MassStorage\msc_sdfat
only this changed: const int chipSelect = 6; 
[image]
Output of Arduino ide:
Sketch uses 27148 bytes (10%) of program storage space. Maximum is 262144 bytes.
Device       : ATSAMD21x18
Version      : v1.1 [Arduino:XYZ] Jan 15 2022 11:55:19
Address      : 0x0
Pages        : 4096
Page Size    : 64 bytes
Total Size   : 256KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
BOD          : true
BOR          : true
Write 27420 bytes to flash (429 pages)
[==============================] 100% (429/429 pages)
Done in 5.329 seconds
Verify 27420 bytes of flash
[==============================] 100% (429/429 pages)
Verify successful
Done in 0.384 seconds

hathach, 2023-01-18:
- please attached the compile log output of arduino IDE, your is only upload log. You may want to enable compile verbose mode in preference first.
- also please take the picture of your actual hw setup and post it here as well. Sometimes the issue is on the wiring.

akarimiz1, 2023-01-18:
log output:
[sdlog.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/10449591/sdlog.txt)

Arduino SD examples work fine. sdfat examples not working.
I placed writing on PCB. ( I used adafruit PCB QTpy m0)

[image]
[image]

I used another sd card reader: 
https://store.fut-electronics.com/products/micro-sd-card-module
[image]

again I have problem by msc_sdfat
* there is a point that msc_sd works fine in these scenarios.

hathach, 2023-01-19:
this is not QTPY m0 but a custom board. Unfortunately, We don't have the hardware to reproduce and/or troubleshoot the issue. If your board has compatible issue with sdfat, maybe you could try to run a simpler example from SdFat, double check the config and/or enable its debugging log to troubleshoot.

closed 2023-01-19 by hathach