adafruit/Adafruit_TinyUSB_Arduino #250: Compilation fails due to errors in function 'dcd_init'

BorisBrock, 2023-02-08:
### Operating System

Windows 11

### Arduino IDE version

PlatformIO (6.1.6)

### Board

RP2040

### ArduinoCore version

Raspberry Pi RP2040 (1.7.0)

### TinyUSB Library version

1.18.1

### Sketch & compiled Log (as attached txt files)

When creating an new project for the RP2040 with Adafruit Tiny USB, the build fails:

> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
> PLATFORM: Raspberry Pi RP2040 (1.7.0) > Raspberry Pi Pico
> HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
> DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, raspberrypi-swd)
> PACKAGES: 
>  - framework-arduino-mbed @ 3.1.1 
>  - tool-rp2040tools @ 1.0.2 
>  - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
> LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> Found 43 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- Adafruit TinyUSB Library @ 1.18.1
> |   |-- SdFat - Adafruit Fork @ 2.2.1
> |   |   |-- SPI
> |-- Adafruit NeoPixel @ 1.11.0
> |   |-- Adafruit TinyUSB Library @ 1.18.1
> |   |   |-- SdFat - Adafruit Fork @ 2.2.1
> |   |   |   |-- SPI
> Building in release mode
> Compiling .pio\build\pico\lib9b4\Adafruit TinyUSB Library\arduino\ports\rp2040\Adafruit_TinyUSB_rp2040.cpp.o
> Compiling .pio\build\pico\lib9b4\Adafruit TinyUSB Library\portable\raspberrypi\rp2040\dcd_rp2040.c.o
> Archiving .pio\build\pico\libe8f\libAdafruit NeoPixel.a
> Archiving .pio\build\pico\libFrameworkArduinoVariant.a
> Compiling .pio\build\pico\FrameworkArduino\Interrupts.cpp.o
> Compiling .pio\build\pico\FrameworkArduino\Serial.cpp.o
> Compiling .pio\build\pico\FrameworkArduino\Tone.cpp.o
> Compiling .pio\build\pico\FrameworkArduino\USB\PluggableUSBDevice.cpp.o
> Compiling .pio\build\pico\FrameworkArduino\USB\USBCDC.cpp.o
> Compiling .pio\build\pico\FrameworkArduino\USB\USBSerial.cpp.o
> .pio\libdeps\pico\Adafruit TinyUSB Library\src\portable\raspberrypi\rp2040\dcd_rp2040.c: In function 'dcd_init':
> .pio\libdeps\pico\Adafruit TinyUSB Library\src\portable\raspberrypi\rp2040\dcd_rp2040.c:373:55: error: 'PICO_SHARED_IRQ_HANDLER_HIGHEST_ORDER_PRIORITY' undeclared (first use in this function); did you mean 'PICO_SHARED_IRQ_HANDLER_DEFAULT_ORDER_PRIORITY'?
>   373 |   irq_add_shared_handler(USBCTRL_IRQ, dcd_rp2040_irq, PICO_SHARED_IRQ_HANDLER_HIGHEST_ORDER_PRIORITY);
>       |                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
>       |                                                       PICO_SHARED_IRQ_HANDLER_DEFAULT_ORDER_PRIORITY
> .pio\libdeps\pico\Adafruit TinyUSB Library\src\portable\raspberrypi\rp2040\dcd_rp2040.c:373:55: note: each undeclared identifier is reported only once for each function it appears in
> *** [.pio\build\pico\lib9b4\Adafruit TinyUSB Library\portable\raspberrypi\rp2040\dcd_rp2040.c.o] Error 1
> .pio\libdeps\pico\Adafruit TinyUSB Library\src\arduino\ports\rp2040\Adafruit_TinyUSB_rp2040.cpp: In function 'void TinyUSB_Port_InitDevice(uint8_t)':
> .pio\libdeps\pico\Adafruit TinyUSB Library\src\arduino\ports\rp2040\Adafruit_TinyUSB_rp2040.cpp:108:26: error: 'PICO_SHARED_IRQ_HANDLER_LOWEST_ORDER_PRIORITY' was not declared in this scope; did you mean 'PICO_SHARED_IRQ_HANDLER_DEFAULT_ORDER_PRIORITY'?
>   108 |                          PICO_SHARED_IRQ_HANDLER_LOWEST_ORDER_PRIORITY);
>       |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
>       |                          PICO_SHARED_IRQ_HANDLER_DEFAULT_ORDER_PRIORITY
> *** [.pio\build\pico\lib9b4\Adafruit TinyUSB Library\arduino\ports\rp2040\Adafruit_TinyUSB_rp2040.cpp.o] Error 1
> ============================================================================================== [FAILED] Took 14.61 seconds ==============================================================================================

### What happened ?

Compilation fails with the following error:

> .pio\libdeps\pico\Adafruit TinyUSB Library\src\portable\raspberrypi\rp2040\dcd_rp2040.c: In function 'dcd_init':
> .pio\libdeps\pico\Adafruit TinyUSB Library\src\portable\raspberrypi\rp2040\dcd_rp2040.c:373:55: error: 'PICO_SHARED_IRQ_HANDLER_HIGHEST_ORDER_PRIORITY' undeclared (first use in this function); did you mean 'PICO_SHARED_IRQ_HANDLER_DEFAULT_ORDER_PRIORITY'?

### How to reproduce ?

1. Open PlatformIO
2. Create a new project for the RP2040
3. Add the library "Adafruit Tiny USB" via "lib_deps = adafruit/Adafruit TinyUSB Library @ 1.18.1"
4. Build

BorisBrock, 2023-02-08:
PS: I just tested the same thing with an older version of the library: 1.14.4. This builds without any errors. So the problem must have been introduced between 1.14.4 and 1.18.1.

hathach, 2023-02-08:
your core version is too old, try to update core to latest version to work with latest release of this lib.

closed 2023-02-08 by hathach