adafruit/Adafruit_TinyUSB_Arduino #253: TinyUSB MIDI not working on ESP32 S2 anymore

Robinyoh, 2023-02-26:
### Operating System

MacOS

### Arduino IDE version

2.0.3

### Board

Lolin S2 MINI 

### ArduinoCore version

2.0.7

### TinyUSB Library version

1.18.3

### Sketch & compiled Log (as attached txt files)

/*********************************************************************
 Adafruit invests time and resources providing this open source code,
 please support Adafruit and open-source hardware by purchasing
 products from Adafruit!

 MIT license, check LICENSE for more information
 Copyright (c) 2019 Ha Thach for Adafruit Industries
 All text above, and the splash screen below must be included in
 any redistribution
*********************************************************************/

/* This sketch is enumerated as USB MIDI device. 
 * Following library is required
 * - MIDI Library by Forty Seven Effects
 *   https://github.com/FortySevenEffects/arduino_midi_library
 */

#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

// USB MIDI object
Adafruit_USBD_MIDI usb_midi;

// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

// Variable that holds the current position in the sequence.
uint32_t position = 0;

// Store example melody as an array of note values
byte note_sequence[] = {
  74,78,81,86,90,93,98,102,57,61,66,69,73,78,81,85,88,92,97,100,97,92,88,85,81,78,
  74,69,66,62,57,62,66,69,74,78,81,86,90,93,97,102,97,93,90,85,81,78,73,68,64,61,
  56,61,64,68,74,78,81,86,90,93,98,102
};

void setup()
{
#if defined(ARDUINO_ARCH_MBED) && defined(ARDUINO_ARCH_RP2040)
  // Manual begin() is required on core without built-in support for TinyUSB such as mbed rp2040
  TinyUSB_Device_Init(0);
#endif

  pinMode(LED_BUILTIN, OUTPUT);
  
  //usb_midi.setStringDescriptor("TinyUSB MIDI");

  // Initialize MIDI, and listen to all MIDI channels
  // This will also call usb_midi's begin()
  MIDI.begin(MIDI_CHANNEL_OMNI);

  // Attach the handleNoteOn function to the MIDI Library. It will
  // be called whenever the Bluefruit receives MIDI Note On messages.
  MIDI.setHandleNoteOn(handleNoteOn);

  // Do the same for MIDI Note Off messages.
  MIDI.setHandleNoteOff(handleNoteOff);

  Serial.begin(115200);

  // wait until device mounted
  while( !TinyUSBDevice.mounted() ) delay(1);
}

void loop()
{
  static uint32_t start_ms = 0;
  if ( millis() - start_ms > 266 )
  {
    start_ms += 266;
    
    // Setup variables for the current and previous
    // positions in the note sequence.
    int previous = position - 1;
  
    // If we currently are at position 0, set the
    // previous position to the last note in the sequence.
    if (previous < 0) {
      previous = sizeof(note_sequence) - 1;
    }
  
    // Send Note On for current position at full velocity (127) on channel 1.
    MIDI.sendNoteOn(note_sequence[position], 127, 1);
  
    // Send Note Off for previous note.
    MIDI.sendNoteOff(note_sequence[previous], 0, 1);
  
    // Increment position
    position++;
  
    // If we are at the end of the sequence, start over.
    if (position >= sizeof(note_sequence)) {
      position = 0;
    }
  }

  // read any new MIDI messages
  MIDI.read();  
}

void handleNoteOn(byte channel, byte pitch, byte velocity)
{
  // Log when a note is pressed.
  Serial.print("Note on: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}

void handleNoteOff(byte channel, byte pitch, byte velocity)
{
  // Log when a note is released.
  Serial.print("Note off: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}

### What happened ?

Is there been made changes that if I upload the above example code the default serial port is not showing anymore and also no MIDI port present. This was working for a long time, but now it is not anymore. Trying to get my head around this, but the only thing I can think off is that there has been made some changes in Tiny USB library or in "esp32-hal-tinyusb.c" that is been updated 2 weeks ago.

Also tested is on een ESP32 S2 DEV board from Espressif, com ports still works, but no midi port.

After this happens I need to reflash firmware to get the serial port back. Sending simple script like BlinkWithoutDelay works nicely, also website script, the serial port comes back. 

### How to reproduce ?

Sent the example script to a LOLIN S2 Mini or just only the #include <Adafruit_TinyUSB.h> line will make this happen.

Robinyoh, 2023-02-28:
Found the problem, went back to board manager ESP32 from Espressif 2.0.6 and now it is working as before. They have changed something in 2.0.7 what prevented your great library to work correctly. Will make an issue ticket there as well.

hathach, 2023-03-02:
the issue is not readable, are you using the midi_test example, if so just simply mention it. Otherwise attached the sketch as txt file. Also please attached the compiled log from Arduino IDE as txt file as well.

hathach, 2023-03-06:
should be fixed by release 2.0.0. Please give it a try

Robinyoh, 2023-03-06:
Thx @hathach the 2.0.0 version fixed it! Big Happy over here :-)

hathach, 2023-03-06:
thank you for testing and confirmation. Close issue now since it is fixed

closed 2023-03-06 by hathach

Robinyoh, 2023-03-07:
Hi @hathach 

I think there is still a small issue.
I didn't see it before because I tested it with my own button to midi script where there are being send only 1 note at a time and note back from my DAW to a neopixel. But if you load the midi_test.ino where a note sequence is being played and reset my Lolin S2 MINI you got 2 different cases:

1. After reset there is a MIDI port and Serial port but no midi output from the ESP32 S2 MINI. Sending a MIDI note to it from a DAW the ESP hangs and MIDI and Serial port are gone. Reset again and de MIDI and Serial port are back.
2. After reset there is a MIDI port and Serial port but only sending 4 notes to the midi port. After sending a random MIDI note to the ESP after this is happened from a DAW the midi output sequence is triggered running normal to the MIDI output. Reset again brings us back in 1 of this cases.