adafruit/Adafruit_TinyUSB_Arduino #254: TinyUSB MSC not working on ESP32-S3

brentru, 2023-02-28:
### Operating System

MacOS

### Arduino IDE version

1.8.19

### Board

Feather ESP32-S3, Feather ESP32-S3 Reverse TFT

### ArduinoCore version

2.0.7

### TinyUSB Library version

GitHub latest

### Sketch & compiled Log (as attached txt files)

Sketch used: `examples/MassStorage/msc_external_flash.ino`

Upload settings (Note: I have also tried the TinyUSB partition)
[image]

### What happened ?

The feather hangs after uploading this example. The USB does not re-enumerate and an MSC USB drive does not appear.

### How to reproduce ?

Upload `msc_external_flash.ino` to an Adafruit ESP32-S3 Feather such as the new Adafruit ESP32-S3 Reverse TFT.

brentru, 2023-03-01:
May be related https://github.com/espressif/arduino-esp32/issues/7907

hathach, 2023-03-02:
I just did an test with Feather eps32s3 and it seems to work well. There is a few gotcha though
- after uploaded, S3 need manual pressing reset button. It is hw issue from espressif
- msc_external_flash use FAT file system, you will need to select the partiion scheme with FATFS (not SPIFFS as screenshot)
- if flash is not fomartted, you need to format it with SdFat_format in Adafruit_SPIFlash example as mentioned in the sketch https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/MassStorage/msc_external_flash/msc_external_flash.ino#L21 . I will make it clearer with the example

Let me know if doing all of above changes and it still doesn't work for you.

hathach, 2023-03-03:
I am using arduino-esp32 at master, it should already included 2.0.7. Can you try 
- running msc_ramdisk example, it does not use any additional library to check if usb work
- if msc_ramdisk does not work on your machine, maybe try to plug it into other machine e.g windows/linux. 

Meanwhile I will try to use the exact release version of 2.0.7 to test with.

brentru, 2023-03-03:
@hathach Same issue with msc_ramdisk on 2x machines at 2.0.7.

hathach, 2023-03-06:
just switch to branch https://github.com/espressif/arduino-esp32/tree/release/v2.x to test, and it indeed has issue with TinyUSB lib, ESP32 repeatedly reset with both S2 and S3. Apparently there is difference between master and branch 2.x. I will try to llook more closely to find the root cause

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400262ed
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x524
load:0x4004c000,len:0xa70
load:0x40050000,len:0x2958
entry 0x4004c18c
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4002b1be  PS      : 0x00060c33  A0      : 0x8002a5ca  A1      : 0x3ffcc5d0  
A2      : 0x3ffc4550  A3      : 0x3ffcc774  A4      : 0x00060020  A5      : 0x00000000  
A6      : 0x400a0e0c  A7      : 0xffffffff  A8      : 0x3ffc4558  A9      : 0x00000001  
A10     : 0x00000000  A11     : 0x00000002  A12     : 0xfffffc18  A13     : 0x3ffc59f8  
A14     : 0x00060423  A15     : 0x3ffcc754  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0xfffffc18  LEND    : 0x3ffc59f8  LCOUNT  : 0x40027480  

Backtrace: 0x4002b1bb:0x3ffcc5d0 0x4002a5c7:0x3ffcc5f0 0x40029976:0x3ffcc610 0x40083e85:0x3ffcc650 0x400842ae:0x3ffcc670 0x40081d04:0x3ffcc690 0x400843b1:0x3ffcc6b0 0x400848a8:0x3ffcc6d0 0x40085145:0x3ffcc720
```

hathach, 2023-03-06:
@brentru v2.0.0 should fix this issue, please give it a try

brentru, 2023-03-06:
@hathach Thank you for resolving this issue via v2.0.0! Working for me.

closed 2023-03-06 by brentru