adafruit/Adafruit_TinyUSB_Arduino #274: Gamepad extension to DirectInput limits (128buttons, 4 POV hats, 2 sliders, 6 axis)

schnoog, 2023-03-28:
Hi, thanks for this great library.

I created several USB input controls for flight sim. Currently all my projects run on an Arduino Micro Pro which uses the ArduinoJoystickLibrary by MHeironimus.
I want to switch several of them to ESP32S2, especially to use both cores, one to work with different I2C devices (VL53L0X, MCP23017....), and one to communicate with the PC.
My Problem is, that my controllers use more than the numbers of buttons and hats available within your library.

I'm wondering if it would be possible to modify the gamepad part in a way that it allows the usage of all inputs available with Microsofts DirectInput (https://web.archive.org/web/20131006193239/http://msdn.microsoft.com/en-us/windows/hardware/gg487464.aspx ).
which are
up to
-128 buttons, 
-four points of view (POVs), 
-two sliders, 
and one each of X, Y, Z, Rx, Ry, and Rz.<

Would love to see that implemented.

OrhanYigitDurmaz, 2023-09-07:
might be possible with custom device descriptors, but im still looking to how to implement them.

hathach, 2023-11-30:
you can implement it with custom report descriptors

closed 2023-11-30 by hathach

OrhanYigitDurmaz, 2023-12-01:
> Hi, thanks for this great library.
>
> I created several USB input controls for flight sim. Currently all my projects run on an Arduino Micro Pro which uses the ArduinoJoystickLibrary by MHeironimus. I [...]

https://gist.github.com/OrhanYigitDurmaz/a37b410e4875a625a515a4def4397046 <-- done it here, take a look

OrhanYigitDurmaz, 2023-12-01:
> Thanks alot. Will have a look into your example.

Dont be afraid to ask me