adafruit/Adafruit_TinyUSB_Arduino #276: DISCUSS: bring over “FFS” class from Adafruit_NeoPXL8

PaintYourDragon, 2023-04-03:
Before I go making a pull request of something that may or may not be a good idea…

In the Adafruit_NeoPXL8 library, two of the examples (NeoPXL8/VideoMSC and NeoPXL8HRD/VideoMSC) had a lot of duplicate code carried over from TinyUSB mass storage examples and projects, so I put this in its own class to simplify re-use…and then found myself _starting to use it in a number of non-NeoPXL8 projects as well,_ as it’s just kind of handy.

Wondering if this class might best be **moved here** into the Adafruit_TinyUSB_Arduino library (since the class is closely tied to this), or **spun out** into its own separate library/repo (with Adafruit_TinyUSB_Arduino as a dependency).

This is currently in a NeoPXL8 branch, not main…I haven’t merged it there yet specifically because of this discussion. Branch is here:
https://github.com/adafruit/Adafruit_NeoPXL8/tree/pb-flexi-video-examples

Specifically, the files “FFS.h” and “FFS.cpp”. _All this does_ is make a board’s flash filesystem (previously initialized by installing CircuitPython) accessible to Arduino projects without a bunch of #if/#elif/#else/etc code (that’s all hidden in the class instead) and keeps it live over USB. This is NOT for all mass storage situations (no SD, etc.), ONLY the CIRCUITPY drive w/default partitioning…there are some projects where we’d like this usable to “average folks” without a ton of weird code or being too tied to a specific board (as written I believe it works on anything CircuitPython-capable).

The name can be changed if that’s too problematic. It’s supposed to be an acronym for Flash FileSystem, but could have other interpretations, I just thought it was funny.

hathach, 2023-04-04:
I have go through he FFS contents, IMO it is more of a helper on top of USB MSC + SPIFlash, and shouldn't be included in this library which focuses on USB driver. Though since it is very handy to handle Adafruit FS on USB acrsos boards, I would suggest to create its own library maybe called `Adafruit_USBFS` or `Adafruit_FATUSB` (up to you) etc... also we can also easily update/add more boards support in the future.

PaintYourDragon, 2023-04-04:
Good points. Thank you! I’ll split this out into its own library/repo then.

closed 2023-04-04 by PaintYourDragon

ATMakersBill, 2023-04-04:
@PaintYourDragon Just an FYI, we regularly find our projects using CP for configurability when in reality Arduino might be a fine solution.  That will be more true going forward as RP2040 PIO USB Host is available only in Arduino/C++ API and we want to make filters with that.  

As such, if you do start another project, please let me know - we have several attempts at config-file exposure via Arduino and would love to work with you on this.

PaintYourDragon, 2023-04-05:
@ATMakersBill Config files for Arduino was one of the use cases exactly…I’m using it with ArduinoJson in a project or two.
New repo is live with minimal example. Not yet published to Arduino Library Manager until things are finalized.
https://github.com/adafruit/Adafruit_CPFS