adafruit/Adafruit_TinyUSB_Arduino #279: ESP32-S3 not appearing as a MIDI device

algomusic, 2023-04-12:
### Operating System

MacOS

### Arduino IDE version

2.0.4

### Board

ESP32-S3 dev board

### ArduinoCore version

ESP32

### TinyUSB Library version

2.0.3

### Sketch as ATTACHED TXT

#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

// USB MIDI object
Adafruit_USBD_MIDI usb_midi;

// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

// Variable that holds the current position in the sequence.
uint32_t position = 0;

// Store example melody as an array of note values
byte note_sequence[] = {
  74,78,81,86,90,93,98,102,57,61,66,69,73,78,81,85,88,92,97,100,97,92,88,85,81,78,
  74,69,66,62,57,62,66,69,74,78,81,86,90,93,97,102,97,93,90,85,81,78,73,68,64,61,
  56,61,64,68,74,78,81,86,90,93,98,102
};

void setup()
{
#if defined(ARDUINO_ARCH_MBED) && defined(ARDUINO_ARCH_RP2040)
  // Manual begin() is required on core without built-in support for TinyUSB such as mbed rp2040
  TinyUSB_Device_Init(0);
#endif
  pinMode(LED_BUILTIN, OUTPUT);
  
  //usb_midi.setStringDescriptor("TinyUSB MIDI");

  // Initialize MIDI, and listen to all MIDI channels
  // This will also call usb_midi's begin()
  MIDI.begin(MIDI_CHANNEL_OMNI);

  // Attach the handleNoteOn function to the MIDI Library. It will
  // be called whenever the Bluefruit receives MIDI Note On messages.
  MIDI.setHandleNoteOn(handleNoteOn);

  // Do the same for MIDI Note Off messages.
  MIDI.setHandleNoteOff(handleNoteOff);

  Serial.begin(115200);

  // wait until device mounted
  while( !TinyUSBDevice.mounted() ) delay(1);
}

void loop()
{
  static uint32_t start_ms = 0;
  if ( millis() - start_ms > 266 )
  {
    start_ms += 266;
    
    // Setup variables for the current and previous
    // positions in the note sequence.
    int previous = position - 1;
  
    // If we currently are at position 0, set the
    // previous position to the last note in the sequence.
    if (previous < 0) {
      previous = sizeof(note_sequence) - 1;
    }
  
    // Send Note On for current position at full velocity (127) on channel 1.
    MIDI.sendNoteOn(note_sequence[position], 127, 1);
  
    // Send Note Off for previous note.
    MIDI.sendNoteOff(note_sequence[previous], 0, 1);
  
    // Increment position
    position++;
  
    // If we are at the end of the sequence, start over.
    if (position >= sizeof(note_sequence)) {
      position = 0;
    }
  }

  // read any new MIDI messages
  MIDI.read();  
}

void handleNoteOn(byte channel, byte pitch, byte velocity)
{
  // Log when a note is pressed.
  Serial.print("Note on: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}

void handleNoteOff(byte channel, byte pitch, byte velocity)
{
  // Log when a note is released.
  Serial.print("Note off: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}

### Compiled Log as ATTACHED TXT

Sketch uses 269553 bytes (20%) of program storage space. Maximum is 1310720 bytes.
Global variables use 21128 bytes (6%) of dynamic memory, leaving 306552 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port /dev/cu.wchusbserial56292538521
Connecting....
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz

### What happened ?

The code compiles and uploads, but the Microcontroller does not appear on my computer as a USB MIDI device. No data is able to be transmitted. 

Similar code using a similar library at https://github.com/chegewara/EspTinyUSB does appear as a MIDI device but that library has other issues that MIDI message send and receive are not (yet) working.

### How to reproduce ?

1. Connect the ESP32-S3 dev board to the computer via the 'com' USB port and compile and upload the code.
2. Shift the USB connection to the 'usb' USB port and reset the ESP32-S3. It should appear in the OS (and other apps) on the computer as a MIDI device, as it does with other boards such as the ESP32 and ESP32-S2 that I have tested.

### Debug Log

N/A

### Screenshots

N/A

hathach, 2023-04-14:
are you using the stock midi_test sketch, if yes then just put links or name the example. Otherwise, attach your sketch as txt file.

algomusic, 2023-04-28:
I'm using the stock midi_test sketch from the examples in Adafruit TinyUSB Library -> MIDI.

Since posting the initial issue report, I've tested on two other ESP32 S3 boards, the Lolin S2 Mini and the Lilygo T7 S3. Both display the same results as reported when using the S3 Dev Board.

To repeat, the program compiles and loads, but the board does not show up as a USB MIDI device on my Mac and no MIDI data is transmitted. 

Also reconfirming that an equivalent example from the related library ESPTinyUSD (https://github.com/chegewara/EspTinyUSB/blob/master/examples/device/midi/midi.ino) does appear as a MIDI device, but no MIDI data is transmitted.

rupraph, 2023-06-06:
Hi,

Same issue here on win10. 
I first tested the code on an esp32-s2 lolin mini and I could switch the USB mode to midi, everything detected and working.

Then I tried to reproduce it on an esp32-s3 without success.
My USB 'otg' port remains describded in windows device manager as a "USB JTAG/seriel debug unit.

Also tried exemples to have usb_hid. No mode seems to switch the USB mode. 

My platformio.ini is as it follows:
`
platform = espressif32
board = esp32-s3-devkitc-1-N8R2
framework = arduino

upload_port = COM10
monitor_port = COM10
monitor_speed = 115200

build_flags = 
	-DUSE_TINYUSB 
lib_archive = no

lib_deps = 
	adafruit/Adafruit TinyUSB Library@^2.2.1
	fortyseveneffects/MIDI Library@^5.0.2
`

alf45tar, 2023-07-04:
midi_test sketch works with ArduinoIDE but not with Platformio due to the order of includes.
It is explained here with the workaround https://community.platformio.org/t/tinyusb-definition-errors-on-esp32s3/29382

I am using the following `platform.ini`

```
[platformio]
default_envs = bpi_leaf_s3

[env:bpi_leaf_s3]
platform = espressif32
framework = arduino
board = bpi_leaf_s3

lib_deps =  adafruit/Adafruit TinyUSB Library
	        fortyseveneffects/MIDI Library

build_flags = 	-DUSE_TINYUSB
				-DCFG_TUD_ENABLED
				-DCFG_TUD_MIDI
				'-DCFG_TUSB_CONFIG_FILE="/Users/changeme/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/arduino_tinyusb/include/tusb_config.h"'
```

algomusic, 2023-08-25:
Solved on Arduino IDE. After much trial and error and thanks to a [post](https://community.platformio.org/t/esp32-s3-tinyusb-config-not-working/34209) by ruprup on the platform io community site.

The following changes from the default Arduino IDE setting for S3 boards are required:
- USB CDC On Boot: Enabled
- Upload Mode: "USB OTG CDC (TinyUSB)"
- USB Mode: "USB-OTG (TinyUSB)"

Also, on some boards it may be necessary to:
- Put ESP32-S3 into boot mode to upload the code
- Press RST on ESP after loading

hathach, 2023-08-25:
thank for the update, you should always try with arduino IDE first, then based platformio config based on the defineds of arduino IDE.

closed 2023-08-25 by hathach

shownb, 2024-01-06:
> Solved on Arduino IDE. After much trial and error and thanks to a [post](https://community.platformio.org/t/esp32-s3-tinyusb-config-not-working/34209) by ruprup on the platform io community site. [...]

I'm having the same problem again. Below are my environments.

```bash
arduino-cli compile --fqbn esp32:esp32:esp32s3 ./rawhid --output-dir ./rawhid                                                                                                                                          1 ✘  6s  19:10:19 
Sketch uses 278413 bytes (21%) of program storage space. Maximum is 1310720 bytes.
Global variables use 19904 bytes (6%) of dynamic memory, leaving 307776 bytes for local variables. Maximum is 327680 bytes.

Used library             Version Path
Adafruit TinyUSB Library 2.3.1   /home/noname/Arduino/libraries/Adafruit_TinyUSB_Library
SPI                      2.0.0   /home/noname/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/SPI

Used platform Version Path
esp32:esp32   2.0.14  /home/noname/.arduino15/packages/esp32/hardware/esp32/2.0.14
```

and i used the arduino ide like you set,it’s the same.

any idea is welcome.

aberoev, 2024-02-08:
Also having this problem. USB MIDI device presents itself on a mac but not my windows 10. I am running out of ideas

terrafirma2021, 2024-07-26:
Confirm issue still exists, tried multiple versions. Official examples work without issues

hathach, 2024-09-05:
this is macos specific issue, should be fixed with https://github.com/adafruit/Adafruit_TinyUSB_Arduino/pull/454 using midi_test example. Note that in your sketch, you may need to force re-enumeration like updated example. S3 is a fast chip, macos driver is also very quick. Together s3 board is already enumerated by the time setup() is called to initialize other class driver such as midi

russelh15, 2025-06-13:
Im running into this same issue. I tested on Mac and PC but the Midi example on my ESP32-S3 is not showing up as an audio device