adafruit/Adafruit_TinyUSB_Arduino #292: rp2040 PICO arduino tinyUSB host support

sandman72, 2023-04-26:
**Is your feature request related to a problem? Please describe.**
Unfortunatly PIO based bit-banging USB host support is only available with 120MHz or 240MHz,
'  - CPU Speed must be either 120 or 240 Mhz. Selected via "Menu -> CPU Speed" '  
So it can't be used with Adafruits PICO DVI library which runs at 252MHz.
First tests with the Pi PICO SDK, PICO_DVI library and the full tinyUSB library there seem to work, so it would be nice to use the native tinyUSB Host support instead of the PIO based USB Host support from within the Arduino IDE.

**Describe the solution you'd like**
Make the USB host support in the Adafruit_TinyUSB_Arduino library compatible with the Adafruit PICO DVI library.
Implement Arduino library functions for the existing tiny USB Host support, so that we can use in a keyboard/mouse via an OTG adapter on the Pi PICO Micro-USB port. Use case here: DVI video terminal with the PICO DVI library.  

Optional: different pins for a host USB port instead of the Micro-Usb port, like in the bit-banged host examples, would be nice.

Being able to use an (OTG) USB hub would be a plus, as it would allow to plug in both keyboard and mouse at the same time.

HisashiKato, 2023-04-28:
Adafruit TinyUSB Library for Arduino is rewrite tusb_config_rp2040.h at the location of \libraries\Adafruit_TinyUSB_Library\src\arduino\ports\rp2040\
 as follows, you can use USB Host in RP2040 instead of Pico-Pio-USB USB Host.

```
// Enable device stack
//#define CFG_TUD_ENABLED 1
#define CFG_TUD_ENABLED 0

// Enable host stack with pio-usb if Pico-PIO-USB library is available
/*
#if __has_include("pio_usb.h")
#define CFG_TUH_ENABLED 1
#define CFG_TUH_RPI_PIO_USB 1
#endif
*/
#define CFG_TUH_ENABLED 1
#define CFG_TUH_RPI_PIO_USB 0
```

[image]

For example, you can use it by wiring like this

(Translated from Japanese to English on a translation site)

HisashiKato, 2023-05-08:
Like this?
It is written in Japanese.
http://kato-h.cocolog-nifty.com/khweblog/2023/05/post-74167a.html

HisashiKato, 2023-05-14:
Right now, TinyUSB Library for Arduino needs to rewrite [/src/arduino/ports/rp2040/tusb_config_rp2040.h](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/src/arduino/ports/rp2040/tusb_config_rp2040.h) to enable native host of RP2040.
It would be nice if I could change the enable/disable to the native host in the Arduino sketch without rewriting this.
I think that it is necessary to implement a function of switch something elth.
I tried that, but I couldn't because of my lack of knowledge.
(Translated from Japanese to English on a translation site)

sandman72, 2023-05-15:
I would second on that. Currently a  _#define CFG_TUH_ENABLED 1_  in the Arduino Sketch gets ignored down stream somehow.

As a workaround you can edit the Adafruit_TinyUSB_Arduino\src\arduino\ports\rp2040\tusb_config_rp2040.h directly, which isn't the nicest way, but it works great so far.

The location might differ depending on the installation:
 %homedrive%%homepath%\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.1.1\
      libraries\Adafruit_TinyUSB_Arduino\src\arduino\ports\rp2040\
      ->>> tusb_config_rp2040.h

// Enable device stack
#define CFG_TUD_ENABLED 1

// Enable host stack with pio-usb if Pico-PIO-USB library is available
//#if __has_include("pio_usb.h")
//#define CFG_TUH_ENABLED 1
//#define CFG_TUH_RPI_PIO_USB 1
//#endif

// Enable TinyUSB Host Stack without RPI bit-banging
#define CFG_TUH_ENABLED 1
#define CFG_TUH_RPI_PIO_USB 0

HisashiKato, 2023-05-15:
I have already described it in this post on my blog.
http://kato-h.cocolog-nifty.com/khweblog/2023/05/post-74167a.html
in this way.

```
//--------------------------------------------------------------------
// COMMON CONFIGURATION
//--------------------------------------------------------------------

// 以下の "//#define TUSB_USE_NATIVE_HOST" の "//" を外して "#define TUSB_USE_NATIVE_HOST" にして、
// この "tusb_config_rp2040.h" を保存すると、RP2040内のUSBコントローラーをホストとして有効化してPico-Pio-USBのホストを無効化します。
// 本当なら "#define TUSB_USE_NATIVE_HOST" を、Arduinoのユーザースケッチ(xxx.ino)の冒頭に移したいのですが、
// "#define TUSB_USE_NATIVE_HOST" を、Arduinoのスケッチ側に移すとコンパイルエラーが起きます。
// なにかしらが足りないっぽいのですが、色々試しましたが結局私には分かりませんでした。誰か解決してくれませんか？
//
// If you remove "//" from "//#define TUSB_USE_NATIVE_HOST" below and
// set it to "#define TUSB_USE_NATIVE_HOST" and save this "tusb_config_rp2040.h", 
// the USB controller in RP2040 will be enabled as a host
// and Pico-Pio-USB will be disabled the host.
// I really want to move "#define TUSB_USE_NATIVE_HOST" to 
// the beginning of the Arduino user sketch (xxx.ino), 
// but if I move "#define TUSB_USE_NATIVE_HOST" to the Arduino sketch side,
// a compilation error occurs.
// It seems that I'm missing something, but after trying a lot,
// I just can't figure it out. Can someone please help me out?
//(Translated from Japanese to English on a translation site)

//#define TUSB_USE_NATIVE_HOST

#if !defined(TUSB_USE_NATIVE_HOST)
#define TUSB_NATIVE_HOST 0
#else
#define TUSB_NATIVE_HOST 1
#endif

#if TUSB_NATIVE_HOST == 0

#define CFG_TUSB_RHPORT0_MODE OPT_MODE_DEVICE
// Enable device stack
#define CFG_TUD_ENABLED 1
// Enable host stack with pio-usb if Pico-PIO-USB library is available
#if __has_include("pio_usb.h")
#define CFG_TUH_ENABLED 1
#define CFG_TUH_RPI_PIO_USB 1
#endif

#else

#define CFG_TUSB_RHPORT0_MODE OPT_MODE_HOST
// Disable device stack
#define CFG_TUD_ENABLED 0
// Enable host stack without pio-usb
#define CFG_TUH_ENABLED 1
#define CFG_TUH_RPI_PIO_USB 0

#endif

/*
#define CFG_TUSB_RHPORT0_MODE OPT_MODE_DEVICE

// Enable device stack
#define CFG_TUD_ENABLED 1

// Enable host stack with pio-usb if Pico-PIO-USB library is available
#if __has_include("pio_usb.h")
#define CFG_TUH_ENABLED 1
#define CFG_TUH_RPI_PIO_USB 1
#endif
*/

```

XZCE, 2023-05-20:
+1 for allowing the use of the native Pico USB port as a Host.

I've done the hack here, and got my USB keyboard working quite well (very reliable).  Thank you.

Unfortunately, when I go to use a "USB Gamepad" it gets an error and stalls, from level 2 debug logging:

"Queue EP 81 with 1 bytes ... usbh_edpt_xfer_with_callback 781: ASSERT FAILED"

(An example for gamepads is in file "hid_app.c in the main TinyUSB repo's examples - for Sony DualShock 4)

Earlier logging shows another 1 byte message was sent on EP 81 (no error there) but I guess the response was lost?

I'd be interested to know if anyone else has managed to get a gamepad working on the Pico in native port host mode?

HisashiKato, 2023-05-20:
_"I'd be interested to know if anyone else has managed to get a gamepad working on the Pico in native port host mode?"_

This video was taken of an Adafruit QT Py RP2040 running a native host using an Arduino sketch I wrote.

https://github.com/adafruit/Adafruit_TinyUSB_Arduino/assets/68432957/b9a08884-8e0e-48f3-b27a-0f9c826a9669

I don't know about the error. sorry.

XZCE, 2023-05-21:
> I don't know about the error. sorry.

It was bad hardware, I tried with another gamepad and it works perfectly too..  Thanks for showing me it working, it gave me the push I needed to look for other reasons why it wasn't working,

joelsernamoreno, 2023-07-14:
Hello!

I am taking advantage of this issue as I have a similar problem

I am using the following examples from the library:

- DualRole -> HID -> hid_remapper: this detects my keyboard, but does not show my keystrokes 

- DualRole -> MassStorage -> mcs_file_explorer: My phone detects the data transfer, but the computer does not recognise the storage. I tried with a flash drive and it didn't work either.

Any tricks to fix this?

Thanks

joelsernamoreno, 2023-07-14:
Hi

I have already fixwd the keyboard and USB Host problem thanks to an article from @HisashiKato 

I still need to fix the msc_file_explorer example problem

Thanks :)

PR #353 (2023-12-23): support rp2040 native usb host configuration
support rp2040 native usb host configuration
- add new example Host/Simple/host_device_info
- fix #292

require https://github.com/earlephilhower/arduino-pico/pull/1910
closed 2023-12-23 by hathach