adafruit/Adafruit_TinyUSB_Arduino #296: HID regression in 2.0.2 when using pico pio usb

delan, 2023-04-30:
### Operating System

Windows 10

### Arduino IDE version

PlatformIO 6.1.6

### Board

pico

### ArduinoCore version

earlephilhower/arduino-pico@3.1.1

### TinyUSB Library version

2.0.2+

### Sketch as ATTACHED TXT

delan/usb3sun@04ba656717d7a04db0a5b68266c1fe9469e5f6f5

### Compiled Log as ATTACHED TXT

n/a

### What happened ?

When upgrading the library from 2.0.1 to 2.0.2, tuh_mount_cb is called, but tuh_hid_mount_cb and tuh_hid_report_received_cb are no longer called.

### How to reproduce ?

1. Get two pico boards, and flash [picoprobe-cmsis-v1.0.2](https://github.com/raspberrypi/picoprobe/releases/tag/picoprobe-cmsis-v1.02) to one of them
2. Connect GND to picoprobe GND, SWCLK to picoprobe GP2, SWDIO to picoprobe GP3, GP1 to picoprobe GP4, GP0 to picoprobe GP5, VBUS to picoprobe VBUS, GP4 to USB-A port D+, GP5 to USB-A port D-
3. Check out delan/usb3sun@04ba656717d7a04db0a5b68266c1fe9469e5f6f5 and `git apply` [repro.patch](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/11361300/repro.patch)
4. Follow the instructions in doc/firmware.md *except* the step about tinyusb2.patch (the patch is not needed when CFG_TUSB_DEBUG=3, and the bug report is better if the library is unmodified)
5. Connect picoprobe to computer, and connect to the picoprobe’s cdc serial port
6. Build and upload the firmware (`pio run -t upload`)
7. Connect a keyboard or mouse to the USB-A port

### Debug Log

* Microsoft Wired Keyboard 600 (045E:0750)
    * [keyboard_old.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/11361290/keyboard_old.txt) = 2.0.1
    * [keyboard_new.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/11361291/keyboard_new.txt) = 2.0.2
    * [keyboard_diff.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/11361292/keyboard_diff.txt)
* Microsoft Wheel Mouse Optical 1.1A (045E:0040)
    * [mouse_old.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/11361294/mouse_old.txt) = 2.0.1
    * [mouse_new.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/11361295/mouse_new.txt) = 2.0.2
    * [mouse_diff.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/11361296/mouse_diff.txt)

JonnyHaystack, 2024-01-04:
Having the same issue here with native USB host on RP2040. `tuh_mount_cb()` works as in the example, but the HID callbacks don't work because it seems to always link the weak definitions from `Adafruit_USBH_Host.cpp` instead of being overridden by my own strong definitions.

delan, 2024-03-04:
Thanks @JonnyHaystack! 3.1.0 is still affected, and I can confirm removing those definitions fixes the problem for me.

@hathach, it looks like the addition of TU_ATTR_WEAK in hid_host.h in hathach/tinyusb#1968 was the breaking change, so if we can’t remove those empty definitions in Adafruit_USBH_Host.cpp for some reason, then we need to remove TU_ATTR_WEAK from the tuh_hid_mount_cb declaration (and tuh_hid_umount_cb and tuh_hid_report_received_cb).

pprice, 2024-06-09:
Can confirm this with `3.1.5` + PlatformIO + rp2020 earlephilhower core; removal of `TU_ATTR_WEAK` from `tuh_hid_mount_cb` and `tuh_hid_umount_cb` definitions in `hid_host.h` , i can see hid callbacks being raised. Workarounds appreciated.

nachomonkey, 2024-12-05:
I can confirm this with PlatformIO, RP2040 earlephilhower core, and Adafruit TinyUSB `3.3.4`, using the Pico's native USB port.
I was surprised to find the HID callbacks weren't firing, because I had no problems with them last year.