adafruit/Adafruit_TinyUSB_Arduino #303: cannot find Adafruit_TinyUSB.h on pico

folkertvanheusden, 2023-05-08:
### Operating System

Linux

### Arduino IDE version

platformio

### Board

RPI RP2040

### ArduinoCore version

latest I guess?

### TinyUSB Library version

adafruit/Adafruit TinyUSB Library@^2.2.0

### Sketch as ATTACHED TXT

#include <Adafruit_TinyUSB.h>

void setup() {
}

void loop() {
}

### Compiled Log as ATTACHED TXT

[log.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/11421500/log.txt)

### What happened ?

Can't find include-file.

### How to reproduce ?

pio run

hathach, 2023-05-08:
This is not a bug of this repo, try to ask on platfromio forum for help

closed 2023-05-08 by hathach

folkertvanheusden, 2023-05-08:
Well the library is advertised in the platformio-registry as such, so it IS a bug in the repo.

[https://registry.platformio.org/libraries/adafruit/Adafruit%20TinyUSB%20Library](https://registry.platformio.org/libraries/adafruit/Adafruit%20TinyUSB%20Library)