adafruit/Adafruit_TinyUSB_Arduino #317: nrf52840

pwilkowski, 2023-07-07:
### Operating System

Windows 10

### Arduino IDE version

PlatformIO Core, version 6.1.7

### Board

nrf52840 DK

### ArduinoCore version

n-able-Arduino 0.1.1

### TinyUSB Library version

adafruit/Adafruit TinyUSB Library@^2.2.1

### Sketch as ATTACHED TXT

```cpp
#include "USB.h"

uint8_t const desc_hid_report[] = {TUD_HID_REPORT_DESC_GENERIC_INOUT(64)};

Adafruit_USBD_HID* usb_hid;

uint16_t get_report_callback(uint8_t report_id, hid_report_type_t report_type, uint8_t* buffer, uint16_t reqlen) {
	// not used in this example
	(void)report_id;
	(void)report_type;
	(void)buffer;
	(void)reqlen;
	return 0;
}

// Invoked when received SET_REPORT control request or
// received data on OUT endpoint ( Report ID = 0, Type = 0 )
void set_report_callback(uint8_t report_id, hid_report_type_t report_type, uint8_t const* buffer, uint16_t bufsize) {
	// This example doesn't use multiple report and report ID
	(void)report_id;
	(void)report_type;

	// echo back anything we received from host
	usb_hid->sendReport(0, buffer, bufsize);
}

void USB::init() {
	usb_hid = new Adafruit_USBD_HID(desc_hid_report, sizeof(desc_hid_report), HID_ITF_PROTOCOL_NONE, 2, true);
	usb_hid->setReportCallback(get_report_callback, set_report_callback);
	usb_hid->setPollInterval(10);
	usb_hid->setReportDescriptor(desc_hid_report, sizeof(desc_hid_report));
	usb_hid->setStringDescriptor("Some Device");

	if (usb_hid->begin()) {
		while (!TinyUSBDevice.mounted()) {
			delay(1);
		}
		Serial.println("USB initialized!");
	}
	else {
		Serial.println("USB failed!");
	}
}

void USB::write(uint8_t* data, int len) {
}
```

### Compiled Log as ATTACHED TXT

There is no log as there is no error, usb simple hid in out example simply fails at `if (usb_hid->begin()) {`
and returns false

### What happened ?

I have dug deeper and it seems like this is the problem
`Adafruit_USBD_Device.cpp`

function `bool Adafruit_USBD_Device::addInterface(Adafruit_USBD_Interface &itf) {`

around the code 
```cpp
    if (desc[0] == 0) {
	  Serial.printf("Failx2 %d", desc_end);
      return false;
    }
```
desc_end returns 32
so this is probably some trivial problem or perhaps a lack of define?

### How to reproduce ?

1. Create project in platformio with following config:
```ini
[env:nrf52840_dk]
platform = n-able
board = nrf52840_dk
framework = arduino
monitor_port = COM5
monitor_speed = 115200
lib_deps =
	https://github.com/h2zero/NimBLE-Arduino.git
	adafruit/Adafruit TinyUSB Library@^2.2.1
build_flags =
	-DARDUINO_NRF52_ADAFRUIT=1
	-DCFG_TUSB_OS_INC_PATH="freertos/"
	-DCFG_TUD_ENABLED=1
	-DCFG_TUD_HID=2
	-DCFG_TUSB_RHPORT0_MODE=0x0001
	-DTASK_PRIO_HIGH=4
```

2. Make Cpp/H file named USB.h, USB.cpp, create empty header with simple include 
```cpp
#include <Arduino.h>
#include "Adafruit_TinyUSB.h"
```

3. Copy contents of the file i attached at the beginning to USB.cpp
4. In main setup. Execute `USB::init();`

### Debug Log

`Failx2 32USB failed!`

hathach, 2023-07-07:
There is lot of not supported
- n-able core not supported
- board is not officially supported
- PIO is not supported.

closed 2023-07-07 by hathach

pwilkowski, 2023-07-07:
For sure, however n-able core literally includes tinyusb in it, I did try it before, now im checking updated version. I do not think this is an issue of any of those as simply buffer seems to be empty, so my best bet is that i am missing something?

hathach, 2023-07-07:
For above platform/board/IDE, you should ask them for support. I don't use any of those and won't be able to provide any info.

pwilkowski, 2023-07-07:
Ok so last question, perhaps this may make those things clearer:

Inside function
```cpp
// Add interface descriptor
// - Interface number will be updated to match current count
// - Endpoint number is updated to be unique
bool Adafruit_USBD_Device::addInterface(Adafruit_USBD_Interface &itf) {
  uint8_t *desc = _desc_cfg + _desc_cfg_len;
  Serial.printf("sizes: %d %d\n",_desc_cfg_maxlen, _desc_cfg_len);```

I have placed simple debug, both `_desc_cfg_maxlen` and `_desc_cfg_len` are zero. Where/how can i configure them?
Imo it doesnt have anything to do with board platform or ide.