adafruit/Adafruit_TinyUSB_Arduino #318: Will PL2303 Chipset be supported (next to CP210X and FTDI) ?

iKK001, 2023-07-07:
Trying to read Serial cmds from my 3rd-party-Device by usage of the `Adafruit TinyUSB Library`'s Example `DualRole-->CDC-->serial_host_bridge`:

Unfortunately, it seems that the built-in PL2303 Chipset inside my 3rd-party-Device is not recognised by the Adafruit TinyUSB Library.

If I connect some other Device to the USB-A plug of the `Adafruit Feather RP2040 USB Host` Board, it all works fine and I can passthrough Serial-cmds to the SerialHost and vice-versa.

However, only if I connect my 3rd-party-Device with this PL2303 Chipset from Profilic, then nothing works !

(i.e. the `tuh_cdc_mount_cb` and `tuh_cdc_umount_cb` callbacks remain silent, unfortunately with the PL2303 device !)

How can I get the PL2303 Chipset to work with your library ?? Is there anything I can tweak inside the library code to make it work ?

Diving deeper into the TinyUSB-Library, I saw that there are two Chipsets distinguished:

- CP210X
- FTDI

But unfortunately not the PL2303 Chipset !

Here the output from the `Adafruit TinyUSB Library`'s Example  `device_info_rp2040`:

Why is this device not working for the `serial_host_bridge` Example ??

```
Device attached, address = 1
Device 1: ID 067b:2303
Device Descriptor:
  bLength             18
  bDescriptorType     1
  bcdUSB              0110
  bDeviceClass        0
  bDeviceSubClass     0
  bDeviceProtocol     0
  bMaxPacketSize0     64
  idVendor            0x067b
  idProduct           0x2303
  bcdDevice           0400
  iManufacturer       1     Prolific Technology Inc. 
  iProduct            2     USB-Serial Controller D
  iSerialNumber       0     
  bNumConfigurations  1

```

hathach, 2023-07-07:
since its driver is not supported yet, only ftdi and cp210x are implemented. PR is welcome.

closed 2023-07-07 by hathach

iKK001, 2023-07-07:
Thank you for the confirmation that PL2303 is not supported. 

Since I am far from being a USB expert, is there any way you can give me an educated hint on where to start looking at in your library ?

hathach, 2023-07-07:
you need to read the specs, read the code. Step debugging with ftdi may help with visualize the software flow. Step debugging with PL chip to see what is missing and fill in the missing piece of code.