adafruit/Adafruit_TinyUSB_Arduino #323: Interface Class, Interface SubClass and Num Interfaces

joelsernamoreno, 2023-08-20:
**Is your feature request related to a problem? Please describe.**
Hello!

Yes, this is a problem. For example, an original keyboard usually has Num Interfaces: 2, Interface Class: 3 and Interface SubClass: 1. For example, RPI Pico has this: Num Interfaces: 3, Interface Class: 2 and Interface SubClass: 2.

The first problem is in USB Host, when using the device_info.ino example with an original keyboard, it shows Num Interfaces, Interface Class and Interface SubClass with a value of **0**.

It would also be fun to be able to modify Num Interfaces, Interface Class and Interface SubClass. For example, you read an original keyboard with USB Host, then configure RPI Pico with this information.

For example, to remove the raspberry pi serial number, I added a function in the library:

**_void Adafruit_USBD_Device::setSerialNumber(uint16_t snumber) {
  _desc_device.iSerialNumber = snumber;
}_**

**Describe the solution you'd like**
1.- USB Host has to detect correctly Num Interfaces, Interface Class and Interface SubClass
2.- To be able to modify this on raspberry pi pico

Thanks!

hathach, 2023-08-22:
I don't understand what you mean, if you think it is an problem, pr is welcome.

closed 2023-08-22 by hathach

joelsernamoreno, 2023-08-24:
Hi @hathach 

I'm trying to modify bNumInterfaces, bInterfaceClass and bInterfaceSubClass

For this I have modified the file Adafruit_USBD_Device.cpp

In Adafruit_USBD_Device::addInterface I have configured this:

**tusb_desc_configuration_t *config = (tusb_desc_configuration_t *)_desc_cfg;
config->bNumInterfaces = 1;**

This works and now bNumberInterfaces is 1 instead of 2.

Now I am trying to modify bInterfaceClass and bInterfaceSubClass, for this I added the following, but it didn't work:

**tusb_desc_desc_interface_t *desc_itf = (tusb_desc_interface_t *)desc_itf;
desc_itf->bInterfaceClass = 3;
desc_itf->bInterfaceSubClass = 3;**

Can you give me a hint?