adafruit/Adafruit_TinyUSB_Arduino #328: Communication not working with device in CDC mode

zorgoz, 2023-08-30:
### Operating System

Windows 10

### Arduino IDE version

Arduino 2.2

### Board

Feather RP2040 with USB Type A Host (brand new)

### ArduinoCore version

3.3.2

### TinyUSB Library version

2.2.2

### Sketch as ATTACHED TXT

The DualRole/CDC/serial_host_bridge example installed with the library
https://raw.githubusercontent.com/adafruit/Adafruit_TinyUSB_Arduino/4a93c7d869852c03b198d74cefe87de4ba071eca/examples/DualRole/CDC/serial_host_bridge/serial_host_bridge.ino

### Compiled Log as ATTACHED TXT

[compile-log.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/12474161/compile-log.txt)

### What happened ?

- I have a USB-serial RFID reader
- The device is recognized correctly both by Windows and the device_info.ino
[image]
[device-info.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/12474191/device-info.txt)
- Works with **putty** as expected: it sends a hexadecimal string 
[image]
- When connected to the board with serial_host_bridbe.ino however, it is recognized as CDC device to be connected...
[image]
...but it is unable to transmit anything to the board 99% of the time. Actually it went just OK at the first run, and I really was happy. But I can't really give a list of steps that make it work in a reproducible manner at least once before stopping. The device is beeping that it was unable to communicate. I have tried the same with a Symbol BCR set to serial port emulation. Same result.
- I have tried to set a different baud-rate in SerialHost.begin(rate); but to no avail. On Windows works with quite any rate from 9600 8N1 up. I don't get the same error beep even with choosing other settings and not getting the data of course. So it seems that it is something between the board and the serial device.
- I can reproduce the same error beep from the device if I give only power to it (with a charger) and try to scan a badge
- As it seems that there is no physical debugging possible on this board, I am stuck with printing to the serial monitor. The only thing I see is that SerialHost.available() never results to true.

### How to reproduce ?

Connect a serial device and check

zorgoz, 2023-08-30:
Big update.
I can make it work in a reproducible manner by inserting a delay of at least 50ms before the `SerialHost.begin(115200);`
[image]

However, removing and reinserting the device breaks the functionality.

hathach, 2023-08-31:
hi, thank you for using Adafruit product and report this issue. Since we don't have your device in question, it may take some time to troubleshoot this. I will try my best to suggest way to test/gather info, and you will need to try it out on the hardware. This could take back and forth a few times. So please be patient.

1. Can you provide full configuration descriptor of your device. You can use using this tool https://www.uwe-sieber.de/usbtreeview_e.html and post the descriptor here.
2. Unfortunately, feather usb host does not have room for an swd, and not even with bottom pad. Though we can eanble tinyusb stack log via UART (Serial1). You will need to
   - modify CFG_TUSB_DEBUG=1 or 2 here https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/src/arduino/ports/rp2040/tusb_config_rp2040.h#L52
   - use additional usb2uart e.g ftdi, cp2014 etc and hook up the TX/RX from the Feather usb host at 115200. It should print out log from, CFG_TUSB_DEBUG=2 can be a bit extensive. (I will check to see we could omit the log from device stack later on).

PS: I am on vacation until next Tuesday, so thing may be a bit slow. Thanks.

zorgoz, 2023-09-01:
Here is the device descriptor
[tr-descriptor.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/12495563/tr-descriptor.txt)

I am thinking about how to sniff on the USB communication...

lyusupov, 2023-09-03:
Please, let me to share my own experience...
I've noticed that:
1) when the USB CDC **client device** for the RP2040 USB PIO Host **uses TinyUSB stack** - in most cases everything works nicely ;
2) however, when the USB **client device uses an USB stack other than TinyUSB** ( like [STM32 USB CDC](https://github.com/stm32duino/Arduino_Core_STM32) or [CH9102F CDC](https://datasheet.lcsc.com/lcsc/2108181630_WCH-Jiangsu-Qin-Heng-CH9102F_C2858418.pdf) ) - chances of establishing a proper CDC connection with the RP2040 PIO Host are much lesser.

zorgoz, 2023-09-04:
I am using TinyUSB stack - it won't compile without it.
[image]
By the way: upgrading to the new core version 3.4.0 did not change anything.
Working on grabbing the log.

lyusupov, 2023-09-04:
> I am using TinyUSB stack

It is very likely that your ["Jantar Codeks Table Reader"](https://jantar.si/pdf/JantarTableReaderManual-en.pdf) **USB client device** internal firmware does NOT use TinyUSB stack.

zorgoz, 2023-09-04:
Ah you mean the other device. That's true. But most don't... :(

zorgoz, 2023-09-04:
Please give me some advice, how can I tap into the USB host RX/TX?
It seems that those MCU pins are not mapped to any DuPont pin on the board. The host pins are directly connected to MCU GPIO ports. Or I have misunderstood something.
[image]

zorgoz, 2023-09-04:
I have managed to tap on to Serial1. 
This file is the log I got with a combined version of _device_info_ and _serial_host_bridge_.
[debuglog.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/12514501/debuglog.txt)
The last line appeared when I showed the card to the reader.
This one is the same, with delay. And working as expected.
[debuglog-with-delay.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/12514594/debuglog-with-delay.txt)

This one I got with the original _serial_host_bridge_. 
[debuglog-bare-without-delay.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/12514512/debuglog-bare-without-delay.txt)
Same error:
[image]
This is the original, with delay:
[debuglog-bare-with-delay.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/12514620/debuglog-bare-with-delay.txt)

Note: the error occures only once, and nothing at all is sent to the debug output after that.

One interesting aspect however is, that with CFG_TUSB_DEBUG = 1, the debug output is just that: `get_itf 172: ASSERT FAILED` lots of times.

hathach, 2023-09-06:
I am just back from holidays, thank you for sharing device descriptor and debug log, look like it is just normal ACM device. Maybe device does not response as tinyusb host driver expect in time or something. However the log with CFG_TUSB_DEBUG is a bit difficult to follow since it combine both device and host. I have made an PR to update tinyusb which include sepearte log setting for host and device https://github.com/adafruit/Adafruit_TinyUSB_Arduino/pull/329

Can you try switching to the PR, then disable device stack log by adding `#define CFG_TUD_LOG_LEVEL 0xff` (while keeping CFG_TUSB_DEBUG=2) to the file tusb_config_rp2040.h

zorgoz, 2023-09-12:
I have checked that PR out, and added CFG_TUD_LOG_LEVEL right after CFG_TUSB_DEBUG
[image]
And it won't compile. With or without `ifndef`.
[CFG_TUD_LOG_LEVEL.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/12584955/CFG_TUD_LOG_LEVEL.txt)

zorgoz, 2023-09-12:
I have checked that PR out. and added `#define CFG_TUD_LOG_LEVEL 0xff`.
[image]
But won't compile. With or without `#ifndef`.
[CFG_TUD_LOG_LEVEL.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/12585067/CFG_TUD_LOG_LEVEL.txt)

hathach, 2023-09-12:
sorry my bad, please change the `#define CFG_TUD_LOG_LEVEL 3` (larger than CFG_TUSB_DEBUG=2) 
 and try again, this should disable device log. I will make more update to make 0xff as explicit OFF later on.

zorgoz, 2023-09-12:
I saw that it got merged. Hence I updated to 2.2.2 and changed the header file there.
```
#define CFG_TUSB_DEBUG 2
#define CFG_TUD_LOG_LEVEL 3
```

OK, with delay.
[log.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/12586509/log.txt)
[output.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/12586513/output.txt)

NOK, without delay.
[log.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/12586520/log.txt)

hathach, 2023-09-29:
sorry, I was busy with other works, there is a new update to the tinyusb recently. Please pull and try it out, maybe it could fix something.

zorgoz, 2023-09-30:
Hello,

Seems that with the recent updates, the issue is solved. No delay is needed to work and even the unplug-replug is working.
Thank you for your effort.

hathach, 2023-10-01:
Great, thank you for your report and feedback.

closed 2023-10-01 by hathach