adafruit/Adafruit_TinyUSB_Arduino #342: msc_file_explorer bugs

Stass48, 2023-10-30:
### Operating System

Windows 10

### Arduino IDE version

Arduino IDE v2.2.1

### Board

Raspberry Pi Pico

### ArduinoCore version

3.6.0 (by Earle F. ...)

### TinyUSB Library version

2.2.5

### Sketch as ATTACHED TXT

Like in Example **msc_file_explorer.ino**.

### Compiled Log as ATTACHED TXT

Sketch uses 77556 bytes (3%) of program storage space. Maximum is 2093056 bytes.
Global variables use 25872 bytes (9%) of dynamic memory, leaving 236272 bytes for local variables. Maximum is 262144 bytes.
Resetting COM10
Converting to uf2, output size: 201728, start address: 0x2000
Scanning for RP2040 devices
Flashing L: (RPI-RP2)
Wrote 201728 bytes to L:/NEW.UF2

**(All are OK)**

### What happened ?

If I use the code from the example as is, then outputting text to the port (function Serial.println()) does not work. If I change Serial to Serial1, I can get output from the GP0 pin.

### How to reproduce ?

Just compile Example and flash to Raspberry.

hathach, 2023-10-31:
Will it work if you uncomment and wait for Serial to connect before runniing the code ?

Stass48, 2023-11-08:
**_Serial.begin(115200);
  delay(1000);_**

Are solved this problem :)

closed 2023-11-08 by Stass48