adafruit/Adafruit_TinyUSB_Arduino #357: ESP32-S3 compilation error

JasXSL, 2023-12-30:
### Operating System

Windows 10

### Arduino IDE version

Arduino 2.2.1

### Board

ESP32-S3-USB-OTG

### ArduinoCore version

1.8.6

### TinyUSB Library version

2.3.1

### Sketch as ATTACHED TXT

/*********************************************************************
 Adafruit invests time and resources providing this open source code,
 please support Adafruit and open-source hardware by purchasing
 products from Adafruit!

 MIT license, check LICENSE for more information
 Copyright (c) 2019 Ha Thach for Adafruit Industries
 All text above, and the splash screen below must be included in
 any redistribution
*********************************************************************/

/* This sketch is enumerated as USB MIDI device. 
 * Following library is required
 * - MIDI Library by Forty Seven Effects
 *   https://github.com/FortySevenEffects/arduino_midi_library
 */

#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

// USB MIDI object
Adafruit_USBD_MIDI usb_midi;

// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

// Variable that holds the current position in the sequence.
uint32_t position = 0;

// Store example melody as an array of note values
byte note_sequence[] = {
  74,78,81,86,90,93,98,102,57,61,66,69,73,78,81,85,88,92,97,100,97,92,88,85,81,78,
  74,69,66,62,57,62,66,69,74,78,81,86,90,93,97,102,97,93,90,85,81,78,73,68,64,61,
  56,61,64,68,74,78,81,86,90,93,98,102
};

void setup()
{
#if defined(ARDUINO_ARCH_MBED) && defined(ARDUINO_ARCH_RP2040)
  // Manual begin() is required on core without built-in support for TinyUSB such as mbed rp2040
  TinyUSB_Device_Init(0);
#endif

  //pinMode(LED_BUILTIN, OUTPUT);
  
  //usb_midi.setStringDescriptor("TinyUSB MIDI");

  // Initialize MIDI, and listen to all MIDI channels
  // This will also call usb_midi's begin()
  MIDI.begin(MIDI_CHANNEL_OMNI);

  // Attach the handleNoteOn function to the MIDI Library. It will
  // be called whenever the Bluefruit receives MIDI Note On messages.
  MIDI.setHandleNoteOn(handleNoteOn);

  // Do the same for MIDI Note Off messages.
  MIDI.setHandleNoteOff(handleNoteOff);

  Serial.begin(115200);

  // wait until device mounted
  while( !TinyUSBDevice.mounted() ) delay(1);
}

void loop()
{
  static uint32_t start_ms = 0;
  if ( millis() - start_ms > 266 )
  {
    start_ms += 266;
    
    // Setup variables for the current and previous
    // positions in the note sequence.
    int previous = position - 1;
  
    // If we currently are at position 0, set the
    // previous position to the last note in the sequence.
    if (previous < 0) {
      previous = sizeof(note_sequence) - 1;
    }
  
    // Send Note On for current position at full velocity (127) on channel 1.
    MIDI.sendNoteOn(note_sequence[position], 127, 1);
  
    // Send Note Off for previous note.
    MIDI.sendNoteOff(note_sequence[previous], 0, 1);
  
    // Increment position
    position++;
  
    // If we are at the end of the sequence, start over.
    if (position >= sizeof(note_sequence)) {
      position = 0;
    }
  }

  // read any new MIDI messages
  MIDI.read();  
}

void handleNoteOn(byte channel, byte pitch, byte velocity)
{
  // Log when a note is pressed.
  Serial.print("Note on: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}

void handleNoteOff(byte channel, byte pitch, byte velocity)
{
  // Log when a note is released.
  Serial.print("Note off: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}

### Compiled Log as ATTACHED TXT

d:\_CODE_\_ARDUINO_\libraries\Adafruit_TinyUSB_Library\src\device\usbd.c: In function 'queue_event':
d:\_CODE_\_ARDUINO_\libraries\Adafruit_TinyUSB_Library\src\device\usbd.c:304:7: error: 'tud_event_hook_cb' undeclared (first use in this function); did you mean 'tud_vendor_tx_cb'?
   if (tud_event_hook_cb) tud_event_hook_cb(event->rhport, event->event_id, in_isr);
       ^~~~~~~~~~~~~~~~~
       tud_vendor_tx_cb
d:\_CODE_\_ARDUINO_\libraries\Adafruit_TinyUSB_Library\src\device\usbd.c:304:7: note: each undeclared identifier is reported only once for each function it appears in
d:\_CODE_\_ARDUINO_\libraries\Adafruit_TinyUSB_Library\src\device\usbd.c:304:26: warning: implicit declaration of function 'tud_event_hook_cb'; did you mean 'tud_vendor_tx_cb'? [-Wimplicit-function-declaration]
   if (tud_event_hook_cb) tud_event_hook_cb(event->rhport, event->event_id, in_isr);
                          ^~~~~~~~~~~~~~~~~
                          tud_vendor_tx_cb
d:\_CODE_\_ARDUINO_\libraries\Adafruit_TinyUSB_Library\src\host\usbh.c: In function 'queue_event':
d:\_CODE_\_ARDUINO_\libraries\Adafruit_TinyUSB_Library\src\host\usbh.c:302:7: error: 'tuh_event_hook_cb' undeclared (first use in this function); did you mean 'tud_vendor_tx_cb'?
   if (tuh_event_hook_cb) tuh_event_hook_cb(event->rhport, event->event_id, in_isr);
       ^~~~~~~~~~~~~~~~~
       tud_vendor_tx_cb
d:\_CODE_\_ARDUINO_\libraries\Adafruit_TinyUSB_Library\src\host\usbh.c:302:7: note: each undeclared identifier is reported only once for each function it appears in
d:\_CODE_\_ARDUINO_\libraries\Adafruit_TinyUSB_Library\src\host\usbh.c:302:26: warning: implicit declaration of function 'tuh_event_hook_cb'; did you mean 'tud_vendor_tx_cb'? [-Wimplicit-function-declaration]
   if (tuh_event_hook_cb) tuh_event_hook_cb(event->rhport, event->event_id, in_isr);
                          ^~~~~~~~~~~~~~~~~
                          tud_vendor_tx_cb
d:\_CODE_\_ARDUINO_\libraries\Adafruit_TinyUSB_Library\src\host\usbh.c: In function 'usbh_control_xfer_cb':
d:\_CODE_\_ARDUINO_\libraries\Adafruit_TinyUSB_Library\src\host\usbh.c:700:5: warning: implicit declaration of function 'TU_LOG1_BUF'; did you mean 'TU_LOG_BUF'? [-Wimplicit-function-declaration]
     TU_LOG1_BUF(request, 8);
     ^~~~~~~~~~~
     TU_LOG_BUF

### What happened ?

Sketch fails to compile. Version 2.2.5 works.

### How to reproduce ?

Attempt to compile with ESP-32-S3-USB-OTG and board esp32 2.0.11 by Espressif installed via arduino with dependencies.

defdefred, 2023-12-31:
Same here compiling msc_ramdisk example for s2 with last arduino-cli.

home/wize/Arduino/libraries/Adafruit_TinyUSB_Library/src/device/usbd.c: In function 'queue_event':
/home/wize/Arduino/libraries/Adafruit_TinyUSB_Library/src/device/usbd.c:304:7: error: 'tud_event_hook_cb' undeclared (first use in this function); did you mean 'tud_vendor_tx_cb'?
   if (tud_event_hook_cb) tud_event_hook_cb(event->rhport, event->event_id, in_isr);
       ^~~~~~~~~~~~~~~~~
       tud_vendor_tx_cb
/home/wize/Arduino/libraries/Adafruit_TinyUSB_Library/src/device/usbd.c:304:7: note: each undeclared identifier is reported only once for each function it appears in

Used library             Version Path           
Adafruit TinyUSB Library 2.3.1   /home/wize/Arduino/libraries/Adafruit_TinyUSB_Library
SPI                      2.0.0   /home/wize/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/SPI

Used platform Version Path                    
esp32:esp32   2.0.14  /home/wize/.arduino15/packages/esp32/hardware/esp32/2.0.14
Error during build: exit status 1

giuliomoro, 2024-01-03:
same for me. I did the following to fix it:

```
diff --git a/src/device/usbd.c b/src/device/usbd.c
index 00cd7f1..1826e02 100644
--- a/src/device/usbd.c
+++ b/src/device/usbd.c
@@ -40,6 +40,8 @@
 #include "device/usbd.h"
 #include "device/usbd_pvt.h"

+#include "common/tusb_common.h"^M
+TU_ATTR_WEAK void tud_event_hook_cb(uint8_t rhport, uint32_t eventid, bool in_isr);^M
 //--------------------------------------------------------------------+
 // ESP32 out-of-sync
 //--------------------------------------------------------------------+
diff --git a/src/host/usbh.c b/src/host/usbh.c
index 0c9a364..14e0e7e 100644
--- a/src/host/usbh.c
+++ b/src/host/usbh.c
@@ -26,6 +26,8 @@

 // ESP32 out-of-sync
 #ifdef ARDUINO_ARCH_ESP32
+#include "common/tusb_common.h"^M
+TU_ATTR_WEAK void tuh_event_hook_cb(uint8_t rhport, uint32_t eventid, bool in_isr);^M
 #include "arduino/ports/esp32/tusb_config_esp32.h"
 #endif
```

shownb, 2024-01-06:
@JasXSL @defdefred @giuliomoro 
I'm also using arduino-cli, I have a question for you, here is my environment. 
```bash
arduino-cli compile --fqbn esp32:esp32:esp32s3 ./rawhid --output-dir ./rawhid                                                                                                                                          1 ✘  6s  19:10:19 
Sketch uses 278413 bytes (21%) of program storage space. Maximum is 1310720 bytes.
Global variables use 19904 bytes (6%) of dynamic memory, leaving 307776 bytes for local variables. Maximum is 327680 bytes.

Used library             Version Path
Adafruit TinyUSB Library 2.3.1   /home/noname/Arduino/libraries/Adafruit_TinyUSB_Library
SPI                      2.0.0   /home/noname/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/SPI

Used platform Version Path
esp32:esp32   2.0.14  /home/noname/.arduino15/packages/esp32/hardware/esp32/2.0.14
```
here is my code,
```c++
#include "Adafruit_TinyUSB.h"
uint8_t const desc_hid_report[] =
{
  TUD_HID_REPORT_DESC_GENERIC_INOUT(64)
};
Adafruit_USBD_HID usb_hid;
void setup()
{
  Serial.begin(115200);
  Serial.println("Adafruit TinyUSB HID Generic In Out example");
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);
  usb_hid.setPollInterval(2);
  usb_hid.setReportDescriptor(desc_hid_report, sizeof(desc_hid_report));
  usb_hid.setReportCallback(0, set_report_callback);
  usb_hid.begin();
  Serial.println("begin done");
  while(!TinyUSBDevice.mounted())
    delay(1);
  Serial.println("done");
  digitalWrite(LED_BUILTIN, HIGH);
}
void loop()
{
}
void set_report_callback(uint8_t report_id, hid_report_type_t report_type, uint8_t const* in_data, uint16_t bufsize)
{
  uint8_t out_data[64];
  for(uint8_t i=0; i < 64; i++)
    out_data[i] = in_data[i];   //echo back
  usb_hid.sendReport(0, out_data, sizeof(out_data));
  if(in_data[1])
    digitalWrite(LED_BUILTIN, LOW);
  else
    digitalWrite(LED_BUILTIN, HIGH);
}
```

but I realized that TinyUSBDevice.mounted() will never be true.I don't have a clue now.

defdefred, 2024-01-06:
Sorry, I'm a beginner with TinyUSB ;-)

hathach, 2024-01-11:
please follow the template for readibility
- attached the sketch as txt
- attached the compile log as txt.

your ArduinoCore for esp 1.8.6 is too old, try to upgrade it to latest

giuliomoro, 2024-01-11:
> your ArduinoCore for esp 1.8.6 is too old, try to upgrade it to latest

@hathach I think that version is a mistake in @JasXSL's bug report. Further down, they say:L

> Attempt to compile with ESP-32-S3-USB-OTG and board esp32 2.0.11 by Espressif installed via arduino with dependencies.

I have the same error with the following configuration:

Operating System
MacOs 12.6.4

Arduino IDE version
2.2.2-nightly-20240102

Board
ESP32-S3 DevKit-C-1 v1.0 (in the Arduino IDE I select "Adafruit Feather ESP32-S3 No PSRAM"

ArduinoCore version
esp32 by Espressif Systems 2.0.11

TinyUSB Library version
Adafruit TinyUSB Library by Adafruit 2.3.1

Even running the latest 2.0.14 ArduinoCore for ESP32 (found [here](https://github.com/espressif/arduino-esp32/releases/tag/2.0.14), though it is weirdly not available through the Arduino IDE's board manager) doesn't fix it.

Upon further investigation, this is down to this repo using TinyUSB's new `tud_event_hook_cb()` and `tuh_event_hook_cb()` callbacks which have been added in 0.16.0 (https://github.com/hathach/tinyusb/commit/188fbd8ed9b5e0f5cc3ddb5e89f7afd421fc7c06), while the ESP32 core seems to be shipping with older versions of  TinyUSB. The fixes I suggest [above](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/357#issuecomment-1874811216) allow this library to compile in the meantime, so they are there for your consideration.

hathach, 2024-01-12:
should be fixed, please pull and try again.

closed 2024-01-12 by hathach

giuliomoro, 2024-01-12:
confirmed working for me

PR #361 (2024-01-12): update tinyusb to pr https://github.com/hathach/tinyusb/pull/2412
Change tuh_event_hook_cb, tud_event_hook_cb to weak default implementation, shoul fix #357