adafruit/Adafruit_TinyUSB_Arduino #360: Support USB MIDI HOST

marchingband, 2024-01-08:
Tinyusb already supports usb midi host, I wonder if it would be much trouble to port it to Arduino?

Myself and I imagine many many others would very much appreciate it. ESP32-S3 is a very powerful target for Midi Host!

I am the creator of https://github.com/marchingband/wvr, and I am porting it to the ESP32-S3 ... I would LOVE to add USB MIDI HOST to the board, currently I need to add a XIAO to the back of my PCB to handle it :( I would be more then happy to help test, or in any way that I can.

Thank you!

marchingband, 2024-01-08:
Possibly this repo does much of the heavy lifting already?
https://github.com/rppicomidi/usb_midi_host/tree/main

rppicomidi, 2024-02-06:
@marchingband The usb_midi_host is an application host driver for TinyUSB. As long as the core processor supports full speed bulk transfers, you can just add both Adafruit TinyUSB and usb_midi_host to your project and it should just work. The comments in the driver code say that TinyUSB ESP32 drivers only supports host mode only with a MAX3421E host shield.

keithngroover, 2024-04-20:
Any luck with this @marchingband ?

marchingband, 2024-04-20:
@keithngroover havn't tried it yet!

keithngroover, 2024-04-20:
Gotcha. I'm trying to make something akin to a very fancy midi filter using
an ESP32S3 (https://www.adafruit.com/product/5323) and USB host shield (
https://www.adafruit.com/product/5858). Here's the project page:
http://theglide.com/newthing

keithngroover, 2024-04-22:
> @marchingband The usb_midi_host is an application host driver for TinyUSB. As long as the core processor supports full speed bulk transfers, you can just add both Adafruit TinyUSB and usb_midi_host [...]

I tried that library @rppicomidi but get the error "Please Select USB Stack: Adafruit TinyUSB Host". However, I can't find any option to select that in the current Arduino IDE. 

The host shield works perfect with the "DualRole" examples included with the TinyUSB Host library, but they didn't provide a midi example.

Any ideas? Thanks in advance.

robertoostenveld, 2024-04-27:
You may want to have a look at https://github.com/tttapa/Control-Surface. For me it worked in getting MIDI over BLE for an ESP32-C6. The documentation includes a table with supported combinations in the board and interface: https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d4a/md_pages_MIDI-over-USB.html

rppicomidi, 2024-04-28:
@marchingband Sorry to send you down a blind alley. I have never done any work with any processor on Arduino other than RP2040 (Rapsberry Pi Pico/Pico W). Please tell me what version of Arduino IDE you are using and what board support and libraries work for the DualRole examples with ESP32, and I will take some time to investigate. There is probably something I did that works only with the RP2040 that is blocking you for the ESP32.

marchingband, 2024-05-07:
@rppicomidi The Dual Role examples seem to compile well.

I am using:
Arduino IDE 2.2.1
Esp32 3.0.0-rc
ESP32S3 Dev Module

When I run your example `usb_midi_host_example.ino`
I get the error `Compilation error: #error "Please Select USB Stack: Adafruit TinyUSB Host"`
The options in Tools-> USB Mode are
1) Hardwar4 CDC and JTAG
2) USB-OTG (TinyUSB)

Likewise when I run the TinyUSB example `host_device_info.ino` I get a similar error `Compilation error: #error This example requires usb stack configured as host in "Tools -> USB Stack -> Adafruit TinyUSB Host"`

I do remember on some old version of the IDE there was a `USB Stack` menu item in the Tools menu, but not any more.
I am not sure what this means for your library and the TinyUSB library here, but neither of them seem to like HOST.

FYI you should be able to test this just as well as me, as these are compilation errors. You can just instal these boards and hit compile. I do have an official ESP32-S3 dev board, and an adapter setup that works for MIDI Host with this board, and I am more then happy to test anything that does compile on this hardware.

I am not sure if it is relevant to you, but I do have a working implementation of USB MIDI Host using esp-idf. I would be happy to paste the USB code here if that is helpful, it uses the esp-idf port of TinyUSB, which is a "component" under esp-idf.

Thanks again for your work, I do believe ESP32S3 is a powerful MIDI Host target, lots of folks will want to use this, especially when the new ESP32-P4 is released, there will be a lot of potential for Espressif parts in the music world.

rppicomidi, 2024-05-07:
@marchingband Thanks for the information. I now have the right board package installed and can duplicate the issue. I am happy to try to get this working on my project, but I believe we should not hijack Adafruit's GitHub page for an issue with my project. I have created [this issue](https://github.com/rppicomidi/usb_midi_host/issues/12) for further discussion. It is up to the developers for this project to decide whether to merge in my project or not.

rppicomidi, 2025-08-13:
@hathach This issue is a request for MIDI Host support for ESP32. Do you need a new issue for Arduino support of RP2040 family? If not, do you need a pull request for tusb_config_rp2040.h to enable MIDI Host? It is only 2 lines.
```
// enable MIDI Host
#define CFG_TUH_MIDI (CFG_TUH_DEVICE_MAX)
```