adafruit/Adafruit_TinyUSB_Arduino #365: Errors when compiling the TinyUSB lbrary using version 2.3.3

DonWT, 2024-01-23:
### Operating System

Windows 10

### Arduino IDE version

Arduino 2.2.1

### Board

Arduino Nano ESP32

### ArduinoCore version

Arduino Nano ESP32

### TinyUSB Library version

2.3.3

### Sketch as ATTACHED TXT

#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

// USB MIDI object
Adafruit_USBD_MIDI usb_midi;

// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, USB_MIDI);

void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);

  USB_MIDI.begin();
}

void loop() {
  USB_MIDI.sendNoteOn(69, 127, 1);
  digitalWrite(LED_BUILTIN, HIGH);  // turn the LED on
  delay(1000);      
  USB_MIDI.sendNoteOff(69, 127, 1);      
  digitalWrite(LED_BUILTIN, LOW);   // turn the LED off
  delay(1000);
}

### Compiled Log as ATTACHED TXT

c:\Users\Don\Arduino\libraries\Adafruit_TinyUSB_Library\src\class\cdc\cdc_host.c:107:49: error: 'CFG_TUH_CDC_FTDI_VID_PID_LIST' undeclared here (not in a function); did you mean 'CFG_TUH_CDC_FTDI'?
 static uint16_t const ftdi_vid_pid_list[][2] = {CFG_TUH_CDC_FTDI_VID_PID_LIST};
                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                 CFG_TUH_CDC_FTDI
c:\Users\Don\Arduino\libraries\Adafruit_TinyUSB_Library\src\class\cdc\cdc_host.c:122:51: error: 'CFG_TUH_CDC_CP210X_VID_PID_LIST' undeclared here (not in a function); did you mean 'CFG_TUH_CDC_CP210X'?
 static uint16_t const cp210x_vid_pid_list[][2] = {CFG_TUH_CDC_CP210X_VID_PID_LIST};
                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                   CFG_TUH_CDC_CP210X
c:\Users\Don\Arduino\libraries\Adafruit_TinyUSB_Library\src\class\cdc\cdc_host.c:122:51: error: initializer element is not constant
c:\Users\Don\Arduino\libraries\Adafruit_TinyUSB_Library\src\class\cdc\cdc_host.c:122:51: note: (near initialization for 'cp210x_vid_pid_list[0][0]')
c:\Users\Don\Arduino\libraries\Adafruit_TinyUSB_Library\src\class\cdc\cdc_host.c:137:50: error: 'CFG_TUH_CDC_CH34X_VID_PID_LIST' undeclared here (not in a function); did you mean 'CFG_TUH_CDC_CH34X'?
 static uint16_t const ch34x_vid_pid_list[][2] = {CFG_TUH_CDC_CH34X_VID_PID_LIST};
                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                  CFG_TUH_CDC_CH34X
c:\Users\Don\Arduino\libraries\Adafruit_TinyUSB_Library\src\class\cdc\cdc_host.c:137:50: error: initializer element is not constant
c:\Users\Don\Arduino\libraries\Adafruit_TinyUSB_Library\src\class\cdc\cdc_host.c:137:50: note: (near initialization for 'ch34x_vid_pid_list[0][0]')

### What happened ?

The sketch compiles clean and runs against the 2.3.2 version of the Adafruit TinyUSB Library but the updated Adafruit library in version 2.3.3 fails with the above error messages.

Reverting back to 2.3.2 and everything is working again.

### How to reproduce ?

Attempt to compile the above sketch against the 2.3.3 version of the library and see the compiler error messages.

Shonky-Donkey, 2024-01-25:
I spent about 3 hours today trying to figure out this same issue. I get the same thing with the "midi_test" example code supplied with the TinyUSB library.

hathach, 2024-01-25:
should be fixed by #367, current ci is setup to build with esp32 v3 (alpha), and didn't catch these out of sync error.

PR #367 (2024-01-25): fix esp32 out of sync with esp core v2.0.14
fix #365
closed 2024-01-25 by hathach