adafruit/Adafruit_TinyUSB_Arduino #379: USB HID GENERIC IN OUT not working anymore since v2.4.0 and later

FeuerSturm, 2024-02-07:
### Operating System

Windows 11

### Arduino IDE version

2.3.0

### Board

Raspberry Pi Pico

### ArduinoCore version

 arduino-pico 3.7.0

### TinyUSB Library version

from 2.40 to 3.0.0 release

### Sketch as ATTACHED TXT

[test_sketch.ino.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/14196832/test_sketch.ino.txt)

### Compiled Log as ATTACHED TXT

no compilation errors

### What happened ?

Starting with release 2.4.0 of Adafruit TinyUSB my project, a Raspberry Pi Pico as ARGB Controller that receives USB RAW HID packets and colors the leds accordingly, stopped working.

- No HID interface/endpoint is created, the Pico doesn't accept any HID data that's been sent to it
- setManufacturerDescriptor is not working anymore either, manufacturer is shown as "UNKNOWN"

Only way to get my project working again is downgrading to release 2.3.3, with that version everything is working as expected.

### How to reproduce ?

Compile and upload the attached test sketch with v2.3.3 check the functionality and then do the same with v2.4.0 or later.

### Screenshots

Manufacturer can be set without problems with v2.3.3:
[image]

NOT working anymore with v2.4.0 and later:
[image]

gepatto, 2024-02-15:
I have a similar issue with a Pico (official and rp2040 zero) and a midi sketch.
v2.3.0 is working but anything later and the usb midi Interface endpoints  (  Audio/Control and Audio/Streaming)
are missing   ( checked on a Mac with USB-Prober )

gepatto, 2024-02-16:
I found that the `setStringDescriptor()`  in my sketch  was causing the error. 
Once I comment out that line  everything works again.

_Not sure if setting a descriptor is now deprecated?_

FeuerSturm, 2024-02-16:
Commenting out `setStringDescriptor` indeed fixed my problems as well, thanks for the hint @gepatto !

hathach, 2024-02-20:
I can't reproduce with 3.0.0 and master. Issue is probably fixed in 3.0.0 or so.

closed 2024-02-20 by hathach

ZetaPhoenix, 2024-02-21:
I am still seeing this error with 3.1.0 and IDE 2.3.2.

Commenting out `setStringDescriptor()` will allow the interfaces to show up.

hathach, 2024-02-22:
> I am still seeing this error with 3.1.0 and IDE 2.3.2.
> 
> Commenting out `setStringDescriptor()` will allow the interfaces to show up.

can you attach your simplest sketch that causes the issue ?

hathach, 2024-02-22:
test againn with stock hid_generic_inout and test sketch in the first post, both are working well.

gepatto, 2024-02-22:
Here's my minimal example code:
```
#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

Adafruit_USBD_MIDI usb_midi;
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

void setup() {
#if defined(ARDUINO_ARCH_MBED) && defined(ARDUINO_ARCH_RP2040)
  // Manual begin() is required on core without built-in support for TinyUSB such as mbed rp2040
  TinyUSB_Device_Init(0);
#endif
  usb_midi.setStringDescriptor("TinyUSB_MIDI");
  MIDI.begin(MIDI_CHANNEL_OMNI);
  Serial.begin(115200);
  while (!TinyUSBDevice.mounted()) delay(1);
  Serial.print("init complete");
}

void loop() {}
```

the line : `usb_midi.setStringDescriptor("TinyUSB_MIDI");`
will cause the midi interface to disappear

In the first image you see what usb prober shows when that line is commented out.
In the second image you can see what happens when I add that line back again. -> 2 interfaces /endpoints are missing

[image]
[image]

Arduino IDE 2.3.2
Master branch Adafruit_TinyUSB_Arduino (date 22 feb 2024)
Macos 14.3.1
Raspberry Pico

ZetaPhoenix, 2024-02-22:
I will add that I built with Ubuntu 22.04 LTS desktop and I am also using a RP2040 Pico.

ZetaPhoenix, 2024-02-22:
I do have a interesting data point to add. If I build `hid_generic_inout` on Windows 11 and then move the device onto the Linux system, I do see the HID interface along with the CDC endpoint. It seems like this could be related to building on Linux/MacOS.

hathach, 2024-03-05:
> Here's my minimal example code:
> 
> ```
> #include <Arduino.h>
> #include <Adafruit_TinyUSB.h>
> #include <MIDI.h>
> 
> Adafruit_USBD_MIDI usb_midi;
> MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);
> 
> void setup() {
> #if defined(ARDUINO_ARCH_MBED) && defined(ARDUINO_ARCH_RP2040)
>   // Manual begin() is required on core without built-in support for TinyUSB such as mbed rp2040
>   TinyUSB_Device_Init(0);
> #endif
>   usb_midi.setStringDescriptor("TinyUSB_MIDI");
>   MIDI.begin(MIDI_CHANNEL_OMNI);
>   Serial.begin(115200);
>   while (!TinyUSBDevice.mounted()) delay(1);
>   Serial.println("init complete");
> }
> 
> void loop() {}
> ```
> 
> the line : `usb_midi.setStringDescriptor("TinyUSB_MIDI");` will cause the midi interface to disappear

work just fine with 3.1.0, "TinyUSB_MIDI" show up in lsusb and midi device is regconized. Please update your library and rp2040 core. 

```
 Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              5 TinyUSB_MIDI
      AudioControl Interface Descriptor:
```

gepatto, 2024-03-05:
Maybe it's a platform specific compile thing.
You're example does not work when compiled on a Mac M1 , 
Arduino IDE 2.3.2
Board:  Raspberry Pi Pico/Rp2040 : 3.7.2
Tiny_USB: 3.1.0
Is there anything that needs updating?

There are no midi endpoints! (when setStringDescriptor is used , without it it's fine )

hathach, 2024-03-06:
yours is all update, it can be either due to compiled file or macos/M1 specific issue. I mostly work on Linux. Here is my uf2 [issue379.ino.zip](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/14503904/issue379.ino.zip), if possible, you could attached yours here as well, I will try it out on my machine

gepatto, 2024-03-06:
Your uf2 works.
Here's mine (your code , different name)
[minimalmidi.ino.uf2.zip](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/14512572/minimalmidi.ino.uf2.zip

hathach, 2024-03-13:
> Your uf2 works. Here's mine (your code , different name) [minimalmidi.ino.uf2.zip](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/14512572/minimalmidi.ino.uf2.zip

your uf2 does not show up as MIDI device under my PC. Therefore I think it is your toolchain setup. Can you provide your compile log as txt file here

gepatto, 2024-03-13:
[compilelog.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/14592095/compilelog.txt)

hathach, 2024-03-14:
> [compilelog.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/14592095/compilelog.txt) working
> 
> [compilelog2.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/14592175/compilelog2.txt) not working

these seems to be the log for a more complicated sketch. Can you provide the log that is compiled with this simple sketch. which we are testing with

```C
#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

Adafruit_USBD_MIDI usb_midi;
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

void setup() {
#if defined(ARDUINO_ARCH_MBED) && defined(ARDUINO_ARCH_RP2040)
  // Manual begin() is required on core without built-in support for TinyUSB such as mbed rp2040
  TinyUSB_Device_Init(0);
#endif
  usb_midi.setStringDescriptor("TinyUSB_MIDI");
  MIDI.begin(MIDI_CHANNEL_OMNI);
  Serial.begin(115200);
  while (!TinyUSBDevice.mounted()) delay(1);
  Serial.println("init complete");
}

void loop() {}
```C

gepatto, 2024-03-14:
[compilelog.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/14606980/compilelog.txt)

here it is (above sketch was save as minimalmidi.ino)

hathach, 2024-03-21:
the log seems to be OK, it probably due to caching issue when there is a huge changes from TinyUSB v2 -> v3. Would you 
- remove the TinyUSB v2 in the core installation folder with the v3 verion 

then try again, if the issue still exists, I will need an tinyusb core log, you can eanble it by changing
- CFG_TUSB_DEBUG = 2 here https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/src/arduino/ports/rp2040/tusb_config_rp2040.h#L68
- then attached an usb2uart to pin tx/rx to the pico since tinyusb core log is dump via Serial1.

gepatto, 2024-03-22:
I replaced all files from the /Users/developer/Library/Arduino15/packages/rp2040/hardware/rp2040/3.7.2/libraries/Adafruit_TinyUSB_Arduino directory (which seemed to be v2.3.0)  with the files from the 3.1.1 library (installed through the library manager)  and now it works.
I didn't realize before that there was another TinyUSB library installed in the rp2040 core directory.  

Thanks for all the help!

hathach, 2024-03-22:
Thank you for confirmation, it is indeed confusing, evenn with me. I am glad we sort it out though.