adafruit/Adafruit_TinyUSB_Arduino #380: (MSC) setStartStopCallback -- callback is never called

acscd, 2024-02-08:
### Operating System

Linux

### Arduino IDE version

1.8.19

### Board

Raspberry Pi Pico

### ArduinoCore version

arduino-pico 3.0.7

### TinyUSB Library version

2.3.0

### Sketch as ATTACHED TXT

[msc_sdcard_test.ino.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/14203947/msc_sdcard_test.ino.txt)

### Compiled Log as ATTACHED TXT

[log.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/14203990/log.txt)

### What happened ?

In the attached sketch, I set a startStop callback with setStartStopCallback (so the drive can be made to disappear like it should upon eject), but it is never called. Attempting to eject the mass storage device fails with an error message and the device remains available.

### How to reproduce ?

1. run sketch
2. attempt to eject drive
3. message from callback does not appear
4. computer gives error message, device remains available

### Screenshots

[image]

hathach, 2024-02-20:
Liunx may not send out start/stop command as part of eject procedure. Try out windows to see if it does not trigger callback. Feel free to re-open issue if you are sure you see the START/STOP command in the bus data, and callback still is not invoked.

closed 2024-02-20 by hathach

acscd, 2024-02-21:
My bad, I did a little probing and it looks like Linux sends a Prevent/Allow Medium Removal command instead. Should I open a feature request on base TinyUSB about support for that command?

[usb log2.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/14353177/usb.log2.txt)