adafruit/Adafruit_TinyUSB_Arduino #389: examples of TinyUSB 3.0.0 do not compile on Arduino-IDE 2.3.1

beta-tester, 2024-02-19:
### Operating System

Linux

### Arduino IDE version

Arduino IDE 2.3.1 (tried as AppImage and also as ZIP on Ubuntu 23.10)

### Board

[Waveshare ESP32-S3 Zero](https://www.waveshare.com/esp32-s3-zero.htm)

### ArduinoCore version

1.8.6 (not sure if this is the value of the ArduinoCore version)

### TinyUSB Library version

3.0.0

### Sketch as ATTACHED TXT

[Adafruit TinyUSB Arduino | Examples | HID | hid_dual_interfaces](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/examples/HID/hid_dual_interfaces)

### Compiled Log as ATTACHED TXT

```
In file included from /home/user/Arduino/libraries/Adafruit_TinyUSB_Library/src/Adafruit_TinyUSB.h:68,
                 from /tmp/.arduinoIDE-unsaved2024119-7116-1kwzfct.fann/hid_dual_interfaces/hid_dual_interfaces.ino:12:
/home/user/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/video/Adafruit_USBD_Video.h:41:15: error: 'tusb_desc_video_control_camera_terminal_t' has not been declared
   addTerminal(tusb_desc_video_control_camera_terminal_t const *camera_terminal);
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/video/Adafruit_USBD_Video.h:43:15: error: 'tusb_desc_video_control_output_terminal_t' has not been declared
   addTerminal(tusb_desc_video_control_output_terminal_t const *output_terminal);
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/video/Adafruit_USBD_Video.h:43:3: error: 'bool Adafruit_USBD_Video::addTerminal(const int*)' cannot be overloaded with 'bool Adafruit_USBD_Video::addTerminal(const int*)'
   addTerminal(tusb_desc_video_control_output_terminal_t const *output_terminal);
   ^~~~~~~~~~~
/home/user/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/video/Adafruit_USBD_Video.h:41:3: note: previous declaration 'bool Adafruit_USBD_Video::addTerminal(const int*)'
   addTerminal(tusb_desc_video_control_camera_terminal_t const *camera_terminal);
   ^~~~~~~~~~~
/home/user/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/video/Adafruit_USBD_Video.h:49:18: error: 'tusb_desc_video_format_uncompressed_t' has not been declared
   bool addFormat(tusb_desc_video_format_uncompressed_t const *format);
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/video/Adafruit_USBD_Video.h:50:17: error: 'tusb_desc_video_frame_uncompressed_continuous_t' has not been declared
   bool addFrame(tusb_desc_video_frame_uncompressed_continuous_t const *frame);
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/video/Adafruit_USBD_Video.h:52:20: error: 'tusb_desc_video_streaming_color_matching_t' has not been declared
   addColorMatching(tusb_desc_video_streaming_color_matching_t const *color);
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/video/Adafruit_USBD_Video.h:61:3: error: 'tusb_desc_video_control_camera_terminal_t' does not name a type; did you mean 'tusb_desc_cs_video_ctl_itf_hdr_t'?
   tusb_desc_video_control_camera_terminal_t _camera_terminal;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   tusb_desc_cs_video_ctl_itf_hdr_t
/home/user/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/video/Adafruit_USBD_Video.h:62:3: error: 'tusb_desc_video_control_output_terminal_t' does not name a type; did you mean 'audio_desc_output_terminal_t'?
   tusb_desc_video_control_output_terminal_t _output_terminal;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   audio_desc_output_terminal_t
/home/user/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/video/Adafruit_USBD_Video.h:66:5: error: 'tusb_desc_video_format_uncompressed_t' does not name a type; did you mean 'tusb_desc_cs_video_frm_uncompressed_t'?
     tusb_desc_video_format_uncompressed_t uncompressed;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     tusb_desc_cs_video_frm_uncompressed_t
/home/user/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/video/Adafruit_USBD_Video.h:67:5: error: 'tusb_desc_video_format_mjpeg_t' does not name a type; did you mean 'tusb_desc_cs_video_frm_mjpeg_t'?
     tusb_desc_video_format_mjpeg_t mjpeg;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     tusb_desc_cs_video_frm_mjpeg_t
/home/user/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/video/Adafruit_USBD_Video.h:71:5: error: 'tusb_desc_video_frame_uncompressed_continuous_t' does not name a type; did you mean 'tusb_desc_cs_video_frm_uncompressed_t'?
     tusb_desc_video_frame_uncompressed_continuous_t uncompressed_cont;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     tusb_desc_cs_video_frm_uncompressed_t
/home/user/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/video/Adafruit_USBD_Video.h:72:5: error: 'tusb_desc_video_frame_mjpeg_continuous_t' does not name a type; did you mean 'tusb_desc_cs_video_frm_mjpeg_t'?
     tusb_desc_video_frame_mjpeg_continuous_t mjpeg;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     tusb_desc_cs_video_frm_mjpeg_t
/home/user/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/video/Adafruit_USBD_Video.h:75:3: error: 'tusb_desc_video_streaming_color_matching_t' does not name a type; did you mean 'tusb_desc_cs_video_stm_itf_out_hdr_t'?
   tusb_desc_video_streaming_color_matching_t _color_matching;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   tusb_desc_cs_video_stm_itf_out_hdr_t

exit status 1

Compilation error: exit status 1
```

### What happened ?

failed compile

### How to reproduce ?

1. install Adafruit_TinyUSB_Library 3.0.0 with all its dependencies as shown
[image]
```
Already installed MIDI Library@5.0.2
Already installed SdFat - Adafruit Fork@2.2.3
Already installed Adafruit NeoPixel@1.12.0
Already installed Adafruit SPIFlash@4.3.4
Downloading Adafruit TinyUSB Library@3.0.0
Adafruit TinyUSB Library@3.0.0
Installing Adafruit TinyUSB Library@3.0.0
Installed Adafruit TinyUSB Library@3.0.0

```

2. select hid_dual_interfaces example:
[Adafruit TinyUSB Arduino | Examples | HID | hid_dual_interfaces](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/examples/HID/hid_dual_interfaces)

3. select "ESP32-S3 Dev Module" and set "USB CDC On Boot:" to "Enabled"
[image]
5. Verify/Compile
6. failed to compile

hathach, 2024-02-20:
duplicated to #387

closed 2024-02-20 by hathach

beta-tester, 2024-02-20:
thank you.
so i have to wait for version greater than 3.0.0 or apply the fix [from merge 388](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/pull/388) by hand, is that right?

krampster, 2024-03-04:
I worked around this by going back to 2.3.2 (after wasting hours). Would be nice to expedite.

hathach, 2024-03-04:
should be fixed by 3.1.0