adafruit/Adafruit_TinyUSB_Arduino #394: panic at `usb.begin()`

beta-tester, 2024-02-20:
### Operating System

Linux

### Arduino IDE version

2.3.2

### Board

ESP32-S3 Dev Module

### ArduinoCore version

esp 2.0.11

### TinyUSB Library version

3.1.0

### Sketch as ATTACHED TXT

https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/HID/hid_dual_interfaces/hid_dual_interfaces.ino

https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/Video/video_capture/video_capture.ino

### Compiled Log as ATTACHED TXT

no issues at compile time nor upload.

### What happened ?

panic at `usb_xxx.begin()`

### How to reproduce ?

1. select example of Adafruit TinyUSB Library: hid_dual_interfaces.ino
2. selected Board: "ESP32-S3 Dev Module", USB CDC On Boot: "Enabled", Core Debug Level: "Verbose"
3. compile sketch and upload.
4. after reset the module, it run into panic and reboots for ever.

i comment out all usb parts in the sketch and enabled them step by step.
the ESP32-S3 shows panic when i enable the `usb_xxx.begin()` line.

i tried two different ESP32-S3 modules.
[Waveshare ESP32-S3-Zero](https://www.waveshare.com/esp32-s3-zero.htm)
[WROOM-1-N16R8 ESP32-S3-DevKitC-1 clone](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html#user-guide-s3-devkitc-1-v1-1-board-front)

both shows the same behavior.

### Debug Log

```
ELF file SHA256: 560bf9401cdca8b3

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42022fce
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4entry 0x403c98d4
[    98][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420228c6  PS      : 0x00060630  A0      : 0x82001cd8  A1      : 0x3fcebd20  
A2      : 0x00000000  A3      : 0x3fc9b6a0  A4      : 0x3fc9b808  A5      : 0x3fc9b708  
A6      : 0x24260401  A7      : 0x24260401  A8      : 0x00000001  A9      : 0x3fcebc30  
A10     : 0x0000009e  A11     : 0xda5ab709  A12     : 0xda5ab709  A13     : 0x00000006  
A14     : 0x00000000  A15     : 0x00000009  SAR     : 0x00000018  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000002  LBEG    : 0x40056f08  LEND    : 0x40056f12  LCOUNT  : 0x00000000  

Backtrace: 0x420228c3:0x3fcebd20 0x42001cd5:0x3fcebd40 0x42001941:0x3fcebd60 0x42003d1e:0x3fcebd90
```
in endless loop

### Screenshots

[image]

hathach, 2024-02-22:
tested with arduino-esp32 2.0.14 and TinyUSB 3.1.0 without any issue, therefore not reproducibled. Try to upgrade your esp32 core and or configure your module correctly.

closed 2024-02-22 by hathach

beta-tester, 2024-02-22:
i have also the issue that Arduino IDE 2.3.2 does not offer the esp32 version 2.0.14. the highest listed version is 2.0.11.
https://github.com/arduino/arduino-ide/issues/2382
maybe this is the cause of the issues i have.

GitHobi, 2024-02-22:
hmmmm. I have installed Arduino IDE 2.3.2 today - using arduino-esp32 (2.0.14) , Arduino_TinyUsb 3.1.0.
I played around with the "hid_generic_inout" sample ... 
I can confirm it compiles and runs - IF "USB CDC on Boot: "Enabled"" - if it is set to "disable" the scatch crashes right after reset (as I had it before).
Is this expected behavior? Why do I need CDC?

beta-tester, 2024-02-22:
> ... IF "USB CDC on Boot: "Enabled"" - if it is set to "disable" the scatch crashes right after reset (as I had it before). Is this expected behavior? Why do I need CDC?

that's what i also observed.
and in the [hid_dual_interfaces.ino](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/HID/hid_dual_interfaces/hid_dual_interfaces.ino) example, only the first USB sub class (usb_keyboard) is working, the second not (usb_mouse) not.

that's the reason, why i "left" Arduino_TinyUSB_Library and switched to the USB examples of arduino-esp32 2.0.14 core package.
1. there is no need to have "CDC" enabled. the only requirement is to set USB Mode to USB-OTG.
2. and UsbKeyboard + UsbMouse example, [KeyboardAndMouseControl.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/KeyboardAndMouseControl/KeyboardAndMouseControl.ino), is working,

so at the moment, for my purpose, i see no need for using Arduino_TinyUSB_Library anymore.