adafruit/Adafruit_TinyUSB_Arduino #395: compile errors if option "USB Firmware MSC On Boot:" is "Enabled"

beta-tester, 2024-02-21:
### Operating System

Linux

### Arduino IDE version

2.3.2

### Board

ESP32-S3 Dev Module

### ArduinoCore version

esp 2.0.11

### TinyUSB Library version

3.1.0

### Sketch as ATTACHED TXT

[hid_dual_interfaces.ino](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/HID/hid_dual_interfaces/hid_dual_interfaces.ino)

### Compiled Log as ATTACHED TXT

[compile error.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/14361955/compile.error.txt)

### What happened ?

if option "USB Firmware MSC On Boot:" is "Enabled", compiling the sketch will fail.

### How to reproduce ?

1. open example sketch "hid_dual_interfaces.ino"
2.  select Board: "ESP32-S3 Dev Module", USB Firmware MSC On Boot: "Enabled", USB Mode: "USB-OTG"
3. compile/verify
4. => compiler error

### Screenshots

[image]

hathach, 2024-02-22:
only set CDC on boot, other MSC/DFU on boot is not supported.

closed 2024-02-22 by hathach

beta-tester, 2024-02-22:
is it possible to bring up a warning with a message that tells that these combination of settings are wrong, instead of just compiler errors from codes that tells a newbie nothing?