adafruit/Adafruit_TinyUSB_Arduino #396: Unhandled Exception on ESP32-S3 / PlatformIO

GitHobi, 2024-02-21:
### Operating System

Linux

### Arduino IDE version

PlatformIO 6.1.13

### Board

ESP32-S3-DevKit-1

### ArduinoCore version

latest

### TinyUSB Library version

Adafruit_TinyUSB_3.1.0

### Sketch as ATTACHED TXT

I'm basically using the ino file from examples/HID/hid_generic_inout
The only thing I modifed is to put the event handlers before "setup" - as otherwise the file would not compile in PlatformIO.

my platformio.ini looks like this:

[env:esp32-s3-devkitm-1]
platform = espressif32
board = esp32-s3-devkitm-1
framework = arduino

build_flags = 
    -O2
    -std=c++11
    -UUSE_TINYUSB

lib_deps =
  adafruit/Adafruit TinyUSB Library @ ^3.1.0

lib_ldf_mode = deep
lib_archive = no

### Compiled Log as ATTACHED TXT

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42022bce
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4202248e  PS      : 0x00060030  A0      : 0x82001b6e  A1      : 0x3fcebe40  
A2      : 0x00000000  A3      : 0x3fc950a4  A4      : 0x3fc94e88  A5      : 0x80000000  
A6      : 0x00060123  A7      : 0x00000000  A8      : 0x00000001  A9      : 0x3fcebe20  
A10     : 0x00000020  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x600c0030  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000002  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x4202248b:0x3fcebe40 0x42001b6b:0x3fcebe60 0x42001f5c:0x3fcebe80 0x420039c6:0x3fcebeb0

### What happened ?

I'm probably missing something vital - but I can compile the scetch without issues. After downloading it it throws a unhandled exception and reboots, and so on. 
Are there any special settings needed to compile / use the scetch with platformio?

### How to reproduce ?

Compile the scetch
Download it to a ESP32-S3-DevKit-1
Press "reset"
will throw exception and restart.

beta-tester, 2024-02-21:
looks pretty the same as my issue #394 on Arduino IDE 2.3.2

hathach, 2024-02-22:
do you have any issue compile stock example with Arduino IDE ?

GitHobi, 2024-02-22:
I'm using usually not Adruiono IDE ...
Stock examples work! 
Also ESP32TinyUSB examples compile and work without problems. So I assume my environment works. 
The Adafruit_TinyUSB samples compile as well (same board settings) - but once usb.begin() is invoked, they crash ...
I need to admit that I'm using not the original Espressiv-ESP32-S3-DevKit ... but this one https://mischianti.org/vcc-gnd-studio-yd-esp32-s3-devkitc-1-clone-high-resolution-pinout-and-specs/. But as said - the samples work.

hathach, 2024-02-22:
which line you are talking about `usb.begin()` in the example ?

hathach, 2024-02-22:
tested and working just fine with stock example with arduino IDE: arduino-esp32 2.0.14 and tinyusb 3.1.0. Please check your platformion setup

closed 2024-02-22 by hathach