adafruit/Adafruit_TinyUSB_Arduino #399: In version 3.1.0 a TinyUSB application refuses to upload on a NanoESP32

DonWT, 2024-02-25:
### Operating System

Windows 10

### Arduino IDE version

Arduino 2.2.1

### Board

Arduino Nano ESP32

### ArduinoCore version

2.0.13

### TinyUSB Library version

3.1.0

### Sketch as ATTACHED TXT

[Minimal_USBMIDI_test.zip](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/14394953/Minimal_USBMIDI_test.zip)

### Compiled Log as ATTACHED TXT

[image]

### What happened ?

The attached sketch compiles, uploads and runs fine on a Nano ESP32 using TinyUSB 2.4.0 but after updating the library to 3.1.0 it compiles but refuses to upload.

After the compile finishes the port is usually disconnected for a brief period and then it usually reconnects and the upload proceeds.  When I compile using the 3.1.0 version of TinyUSB then that re-connection never occurs and the Nano ESP32 refuses to run.  I have tried resetting the board while it is disconnected but that does not help.

After this has happened then I have to reset the Nano ESP32 before I can use it for other sketches.

### How to reproduce ?

1.  Using a Nano ESP32 compile the attached sketch
2. See that it does not regain its connection when the upload begins

(I am using a NanoESP32, this may occur with other ESP32-S3 boards but I have not tested this.)

### Debug Log

N/A

### Screenshots

See Compiled Log above.

hathach, 2024-02-26:
not reproduciable with Adafruit Feather S3 uploading using CDC.

closed 2024-02-26 by hathach

DonWT, 2024-02-26:
Would you please try reproducing my issue using the midi TinyUSB sketch
that I included in my bug report and could you send me a copy of the CDC
sketch that you used on the Feather S3.  If this is a Nano ESP32 problem
then I want to report it to Arduino.

I will also do some testing using an ESP32 S3 devkit board.

Thank you,

Don.

hathach, 2024-02-26:
I tested with stock msc_ramdisk example, if you could reproduce it with stock example, I will give it a try.

DonWT, 2024-02-27:
I have spent a bit more time poking at this problem and I think that my
initial characterisation of the issue was wrong.

It is not that a 3.1.0 version of an application does not upload, it does
upload correctly and the midi or ramdisk part does work as expected but
what is happening is that the serial port used by the Arduino IDE gets
disconnected and does not get reconnected ready to print any serial monitor
messages.

Looking at my device manager while uploading, I see the COM port in use
disappear from the device manager entry "Ports(COM & LPT)" and a new port
does not appear after the upload completes.  So there is no usable serial
port for the serial monitor.  The net result is the application runs the
TinyUSB side of things OK (midi or ramdisk) but there is no serial monitor
working now.  This was not the case with the 2.4.0 version of your library.

I have the same problem with both my midiUSB application and your
msc_ramdisk application.  However, to get anything to show at all on the
serial monitor using the ramdisk application I had to add a delay(2000)
after the Serial.begin(115200) call:
  Serial.begin(115200);
  //while ( !Serial ) delay(10);   // wait for native usb
  delay(2000);
  Serial.println("Adafruit TinyUSB Mass Storage RAM Disk example");

I have to do this for all ESP32 applications and note that while ( !Serial
) delay(10);
does not seem to work for ESP32 boards.

I hope this makes sense,

Don

hathach, 2024-02-27:
it is expected behavior, while booting, the usb vcom is allready enumerated before you can launch monitor therefore the printf() message is lost.

hathach, 2024-02-28:
no, virtual com work normally as it is, you only need to wait for its connection before sending message. uncomment following lines work for me. Please read my above message for reason.
 
```C
while ( !Serial ) delay(10);   // wait for native usb
```

DonWT, 2024-02-29:
Here is my latest test results:

I am going to compile and upload msc_ramdisk from your Examples collection.  I am using the Arduino IDE version 2.2.1, an Arduino Nano ESP32 configured as such in the board manager and version 3.1.0 of the TinyUSB library.  

The only change I have made to your code is to remove the comment on this line, as you suggested:
    while ( !Serial ) delay(10);   // wait for native usb

This is what my Windows 10 Device Manager shows before compiling and uploading msc_ramdisk:

[image]

msc_ramdisk compiles and uploads OK and the ram disk appears in Windows file manager and I can look at the README.TXT file.
However, once the application finished uploading then the Arduino IDE reported the original COM port (COM20) as disconnected and it did not reconnect, nor were there any other ports shown as available.  

Looking at the msc_ramdisk code I don't see how it got past the "while ( !Serial ) delay(10);" statement unless it found some other port to use - there is none shown now in the Device Manager:

[image]

I did notice that I now have a "Universal Serial Bus device" called "TinyUSB Serial" - is this where the Serial.print message was sent to?  If so then I cannot access this from the Arduino IDE.

hathach, 2024-03-04:
not reproduciable with my setup with feather esp32s3. Check your setup or test with other PC and/or boards for further info.

hathach, 2024-03-05:
[image]

DonWT, 2024-03-07:
I have just received a Banana Pi Leaf S3 and I am going to report how this
board behaves when testing the msc_ramdisk example.

Thank you for the copy of your board definitions.  My own definitions are
basically the same as yours - you have a few more options than I have but I
do not think that these are important.  This is what I have for my Banana
Pi:

[image: image.png]

I am using 3.1.0 of your library.

*Case 1*)  When I compile, upload and run msc_ramdisk as distributed with
your library then I do not see the
"Adafruit TinyUSB Mass Storage RAM Disk example"
message on the Serial Monitor and I do not see a disk created in Windows.

*Case 2)* If I uncomment this statement as you suggested:
//  while ( !Serial ) delay(10);   // wait for native usb
then I get the same result - nothing on the Serial Monitor and no disk
created

*Case 3)* If I comment out all of the Serial statements   //
Serial.begin(115200);
//  while ( !Serial ) delay(10);   // wait for native usb

  // Serial.println("Adafruit TinyUSB Mass Storage RAM Disk example");
then the TinyUSB MSC disk is reliably created in Windows and, of course, no
message appears on the Serial Monitor.

When I try the same tests using a Nano ESP32 then I get:

*Case 1)* The TinyUSB MSC disk appears for a few seconds and then
disappears, the IDE throws this message:
Port monitor error: command 'open' failed: Serial port not found. Could not
connect to COM28 serial port.
The upload was successful but the Nano could not reconnect a Serial port.

*Case 2)* Same result as case 1.
*Case 3)* The TinyUSB MSC disk is reliably created and I get the same Port
Monitor not being created message.

Message ID: ***@***.***
com>
>
>