adafruit/Adafruit_TinyUSB_Arduino #400: USB descriptors not correctly updated if Bluetooth stack is initialized before Adafruit_TinyUSB

tttapa, 2024-03-03:
### Operating System

Linux

### Arduino IDE version

Arduino 2.3.2

### Board

Raspberry Pi Pico (FQBN: `rp2040:rp2040:rpipicow:usbstack=tinyusb,ipbtstack=ipv4btcble`)

### ArduinoCore version

https://github.com/earlephilhower/arduino-pico 3.7.2

### TinyUSB Library version

2.3.0 (bundled with core)

### Sketch as ATTACHED TXT

<details>

<summary>Sketch that reproduces the issue (minimal)</summary>

```cpp
#define HID_REPORT_TYPE_INPUT HID_REPORT_TYPE_INPUT_bt
#define HID_REPORT_TYPE_OUTPUT HID_REPORT_TYPE_OUTPUT_bt
#define HID_REPORT_TYPE_FEATURE HID_REPORT_TYPE_FEATURE_bt
#define hid_report_type_t hid_report_type_t_bt
#include <btstack.h>
#undef HID_REPORT_TYPE_INPUT
#undef HID_REPORT_TYPE_OUTPUT
#undef HID_REPORT_TYPE_FEATURE
#undef hid_report_type_t

void power_on_hci() { hci_power_control(HCI_POWER_ON); }
#include <Adafruit_TinyUSB.h>

Adafruit_USBD_MIDI usb_midi;

void setup() {
#if 1 // When 1, USB MIDI does not work, when 0, it does work
  power_on_hci();
#endif
  pinMode(LED_BUILTIN, OUTPUT);
  usb_midi.begin();
}

void loop() { // Blink as a liveness check
  static bool state = false;
  digitalWrite(LED_BUILTIN, state = !state);
  Serial.printf("%s\n", state ? "on" : "off");
  delay(500);
}
```

</details>

### Compiled Log as ATTACHED TXT

[picow-usb-ble-problem.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/14473116/picow-usb-ble-problem.txt)

### What happened ?

**→ Previously reported at https://github.com/earlephilhower/arduino-pico/issues/2022, which contains more information**

If the Bluetooth stack is initialized before calling the `begin` function on any of the `Adafruit_TinyUSB` instances, the descriptors that `Adafruit_TinyUSB` uses don't include the descriptors for any of those instances.

The Pico does show up correctly after forcing the USB bus to rescan (but just resetting the Pico does not help).

### How to reproduce ?

1. Select the `Raspberry Pi Pico W` in the IDE
2. Enable the Bluetooth stack (`Tools` > `IP/Bluetooth Stack`: `"IPv4 + Bluetooth"`)
3. Enable Adafruit TinyUSB as USB stack (`Tools` > `USB Stack`: `"Adafruit TinyUSB"`)
4. Upload the sketch above (which includes an instance of `Adafruit_USBD_MIDI`).
5. Verify that the board does not show up as a MIDI device, there's no hint of MIDI in the descriptors.

Some debugging was already done by earlephilhower in https://github.com/earlephilhower/arduino-pico/issues/2022, where he was able to rule out any problems with the earlephilhower/arduino-pico core specifically, which is why I've opened an issue here.

From the previous issue:

> **earlephilhower**  
> Yup, the Adafruit library of TinyUSB is not signaling the PC to re-read the descriptor when this happens and hence the PC will keep the old one. If you cause the USB bus to rescan, the device does show up properly:  
> [...]  
> So this is a TinyUSB issue most likely. Adafruit is making a new descriptor and informing TinyUSB about it, but for some reason TinyUSB can't tell the PC about it. Might want to ping them with this info, it may be a simple fix...

earlephilhower, 2024-10-29:
Not actually an issue.  See #454.

A long delay to bring up the BT HCI interface lets the host enumerate the USB device listing before the MIDI interface is added.  Need to manually disconnect/reconnect to get it going, like in the examples.

hathach, 2025-05-21:
not the lib issue, as solution is mentioned above. This occurs when there are mutliple large sub-system bt/usb working together.

closed 2025-05-21 by hathach