adafruit/Adafruit_TinyUSB_Arduino #403: RNDIS/USB network example

i-am-shodan, 2024-03-12:
**Is your feature request related to a problem? Please describe.**
The TinyUSB project has an example of USB networking in the examples dir but this isn't available as a class here

**Describe the solution you'd like**
An example of creating a RNDIS device with this library

**Describe alternatives you've considered**
I've tried to create a class from the MIDI implementation but I can't get it working as the interface doesn't appear. This is on an ESP32-S3 which it should work on. 

**Additional context**
No

hathach, 2024-03-13:
we have no plan to add C++ class for rndis, you can use the native tinyusb API. PRs are welcome.

closed 2024-03-13 by hathach

i-am-shodan, 2024-03-13:
Do you have a link to your plan?

hathach, 2024-03-13:
issue & pr list is the plan.