adafruit/Adafruit_TinyUSB_Arduino #410: cant verify correctly error: missing Adafruit_USBD_CDC.h file eventhough i have it.

fabiendostie, 2024-03-23:
### Operating System

MacOS

### Arduino IDE version

2.3.2

### Board

Seeduino Xiao samd21

### ArduinoCore version

1.8.5

### TinyUSB Library version

Adafruit TinyUSB 3.1.1

### Sketch as ATTACHED TXT

[https://github.com/fabiendostie/Adafruit_TinyUSB_Issue/blob/main/Xiao_samd21_midi_HC-sr04.ino](url)

### Compiled Log as ATTACHED TXT

```
In file included from /private/var/folders/k0/hjpsy66n3hg335tq13ylq_jc0000gn/T/arduino/sketches/4F23009D6D78F625C1DD46B58E7A6D24/sketch/ultrasonicMIDI_v2.ino.cpp:1:0:
/Users/lefab/Library/Arduino15/packages/Seeeduino/hardware/samd/1.8.5/cores/arduino/Arduino.h:173:10: fatal error: Adafruit_USBD_CDC.h: No such file or directory
 #include "Adafruit_USBD_CDC.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1
```

### What happened ?

I cant seem to be able to compile my project because of a missing Adafruit_USBD_CDC.h file. The thing is that, the file in question is on my system: /Users/lefab/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/Adafruit_USBD_CDC.h
So I'm a little puzzled.

### How to reproduce ?

Just load the info file and try to verify the sketch.

### Debug Log

I do not know how to produce this

### Screenshots

[image]
[image]
[image]

hathach, 2024-03-23:
Board is not supported, check with board maker for support

closed 2024-03-23 by hathach

fabiendostie, 2024-03-23:
The Xiao samd21? I've seen it in the list, and actually had it working for
a different project using this library.