adafruit/Adafruit_TinyUSB_Arduino #415: Conflicting Types in 3.1.3

coofercat, 2024-04-27:
### Operating System

MacOS

### Arduino IDE version

Arduino 1.8.13

### Board

DollaTek ESP32-S3-DevKitC-1

### ArduinoCore version

1.8.6

### TinyUSB Library version

3.1.3

### Sketch as ATTACHED TXT

examples/hid_boot_mouse.ino (https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/HID/hid_boot_mouse/hid_boot_mouse.ino)

### Compiled Log as ATTACHED TXT

```
/Users/x/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/host/usbh.c:756:6: error: conflicting types for 'tuh_edpt_abort_xfer'
 bool tuh_edpt_abort_xfer(uint8_t daddr, uint8_t ep_addr) {
      ^~~~~~~~~~~~~~~~~~~
/Users/x/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/host/usbh.c:720:15: note: previous implicit declaration of 'tuh_edpt_abort_xfer' was here
               tuh_edpt_abort_xfer(daddr, tu_edpt_addr(epnum, dir));
               ^~~~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board ESP32S3 Dev Module.
```

### What happened ?

In version 3.1.3, I can't compile anything using the library and get the error above. If I revert to the next previous version available (3.1.1), it compiles correctly.

### How to reproduce ?

1. With an (almost) completely fresh Arduino IDE, install Adafruit TinyUSB and its dependencies
2. Compile the `examples/hid_boot_mouse.ino` sketch
3. Observe error
4. Downgrade the Adafruit TinyUSB library to verion 3.1.1
5. Compile - observe it compiles and uploads

jenschr, 2024-04-28:
I can confirm. Tested with Arduino 2.3.2 and Adafruit Feather S3 NoPSRAM, I get the same error.

hathach, 2024-04-29:
BSP 1.8.6 is too old, please update to at least 2.0.14/15

coofercat, 2024-04-29:
Sorry, I'm not sure where 1.8.6 came from - if I look in Boards Manager, the `Arduino ESP32 Boards` version is 2.0.13. I don't have an option for 14/15, but I'll look to add one of those and re-try.

Wycers, 2024-04-29:
Same issue here

hathach, 2024-04-30:
I will look at this soon, though anyone with “same issue” please post your setup detail (check 1st post)

closed 2025-05-21 by hathach