adafruit/Adafruit_TinyUSB_Arduino #416: Failed to compile msc_sd example

vanbwodonk, 2024-05-01:
### Operating System

Linux

### Arduino IDE version

1.8.19

### Board

Raspberry pi pico

### ArduinoCore version

arduino-pico core 3.8.0

### TinyUSB Library version

3.1.3

### Sketch as ATTACHED TXT

/*********************************************************************
 Adafruit invests time and resources providing this open source code,
 please support Adafruit and open-source hardware by purchasing
 products from Adafruit!

 MIT license, check LICENSE for more information
 Copyright (c) 2019 Ha Thach for Adafruit Industries
 All text above, and the splash screen below must be included in
 any redistribution
*********************************************************************/

/* This example expose SD card as mass storage using
 * default SD Library
 */

#include "SD.h"
#include "Adafruit_TinyUSB.h"
#include "SdFat.h"

const int chipSelect = 10;

Adafruit_USBD_MSC usb_msc;

Sd2Card card;
SdVolume volume;

// the setup function runs once when you press reset or power the board
void setup()
{
  // Set disk vendor id, product id and revision with string up to 8, 16, 4 characters respectively
  usb_msc.setID("Adafruit", "SD Card", "1.0");

  // Set read write callback
  usb_msc.setReadWriteCallback(msc_read_cb, msc_write_cb, msc_flush_cb);

  // Still initialize MSC but tell usb stack that MSC is not ready to read/write
  // If we don't initialize, board will be enumerated as CDC only
  usb_msc.setUnitReady(false);
  usb_msc.begin();

  Serial.begin(115200);
  //while ( !Serial ) delay(10);   // wait for native usb

  Serial.println("Adafruit TinyUSB Mass Storage SD Card example");

  Serial.println("\nInitializing SD card...");

  if ( !card.init(SPI_HALF_SPEED, chipSelect) )
  {
    Serial.println("initialization failed. Things to check:");
    Serial.println("* is a card inserted?");
    Serial.println("* is your wiring correct?");
    Serial.println("* did you change the chipSelect pin to match your shield or module?");
    while (1) delay(1);
  }

  // Now we will try to open the 'volume'/'partition' - it should be FAT16 or FAT32
  if (!volume.init(card)) {
    Serial.println("Could not find FAT16/FAT32 partition.\nMake sure you've formatted the card");
    while (1) delay(1);
  }
  
  uint32_t block_count = volume.blocksPerCluster()*volume.clusterCount();

  Serial.print("Volume size (MB):  ");
  Serial.println((block_count/2) / 1024);

  // Set disk size, SD block size is always 512
  usb_msc.setCapacity(block_count, 512);

  // MSC is ready for read/write
  usb_msc.setUnitReady(true);
}

void loop()
{
  // nothing to do
}

// Callback invoked when received READ10 command.
// Copy disk's data to buffer (up to bufsize) and
// return number of copied bytes (must be multiple of block size)
int32_t msc_read_cb (uint32_t lba, void* buffer, uint32_t bufsize)
{
  (void) bufsize;
  return card.readBlock(lba, (uint8_t*) buffer) ? 512 : -1;
}

// Callback invoked when received WRITE10 command.
// Process data in buffer to disk's storage and 
// return number of written bytes (must be multiple of block size)
int32_t msc_write_cb (uint32_t lba, uint8_t* buffer, uint32_t bufsize)
{
  (void) bufsize;
  return card.writeBlock(lba, buffer) ? 512 : -1;
}

// Callback invoked when WRITE10 command is completed (status received and accepted by host).
// used to flush any pending cache.
void msc_flush_cb (void)
{
  // nothing to do
}

### Compiled Log as ATTACHED TXT

In file included from /home/drw/.arduino15/packages/rp2040/hardware/rp2040/3.8.0/libraries/SD/src/SD.h:24,
                 from /home/drw/Arduino/msc_sd/msc_sd.ino:16:
/home/drw/.arduino15/packages/rp2040/hardware/rp2040/3.8.0/libraries/SDFS/src/SDFS.h: In member function 'virtual int sdfs::SDFSFileImpl::availableForWrite()':
/home/drw/.arduino15/packages/rp2040/hardware/rp2040/3.8.0/libraries/SDFS/src/SDFS.h:279:31: error: 'using element_type = class File32' {aka 'class File32'} has no member named 'availableSpaceForWrite'; did you mean 'availableForWrite'?
  279 |         return _opened ? _fd->availableSpaceForWrite() : 0;
      |                               ^~~~~~~~~~~~~~~~~~~~~~
      |                               availableForWrite
/home/drw/Arduino/msc_sd/msc_sd.ino: At global scope:
msc_sd:24:1: error: 'Sd2Card' does not name a type; did you mean 'SdCard'?
   24 | Sd2Card card;
      | ^~~~~~~
      | SdCard
msc_sd:25:1: error: 'SdVolume' does not name a type; did you mean 'FsVolume'?
   25 | SdVolume volume;
      | ^~~~~~~~
      | FsVolume
/home/drw/Arduino/msc_sd/msc_sd.ino: In function 'void setup()':
msc_sd:48:9: error: 'card' was not declared in this scope
   48 |   if ( !card.init(SPI_HALF_SPEED, chipSelect) )
      |         ^~~~
msc_sd:58:8: error: 'volume' was not declared in this scope; did you mean 'FsVolume'?
   58 |   if (!volume.init(card)) {
      |        ^~~~~~
      |        FsVolume
msc_sd:58:20: error: 'card' was not declared in this scope
   58 |   if (!volume.init(card)) {
      |                    ^~~~
msc_sd:63:26: error: 'volume' was not declared in this scope; did you mean 'FsVolume'?
   63 |   uint32_t block_count = volume.blocksPerCluster()*volume.clusterCount();
      |                          ^~~~~~
      |                          FsVolume
/home/drw/Arduino/msc_sd/msc_sd.ino: In function 'int32_t msc_read_cb(uint32_t, void*, uint32_t)':
msc_sd:86:10: error: 'card' was not declared in this scope
   86 |   return card.readBlock(lba, (uint8_t*) buffer) ? 512 : -1;
      |          ^~~~
/home/drw/Arduino/msc_sd/msc_sd.ino: In function 'int32_t msc_write_cb(uint32_t, uint8_t*, uint32_t)':
msc_sd:95:10: error: 'card' was not declared in this scope
   95 |   return card.writeBlock(lba, buffer) ? 512 : -1;
      |          ^~~~
Multiple libraries were found for "SdFat.h"
 Used: /home/drw/Arduino/libraries/SdFat_-_Adafruit_Fork

### What happened ?

Sd2Card and SdVolume not include in library. Already download and include adafruit fork sdfat.

### How to reproduce ?

1. Open File > Examples > Adafruit TinyUSB Library > DualRole > msc_file_explorer
2. Download and include Adafruit SdFat library
3. Build

hathach, 2025-05-21:
should be fixed now with latest Adafruit SdFat fork release

closed 2025-05-21 by hathach

leahanderson1, 2025-07-14:
still getting this error, macos
```
/private/var/folders/fm/30cx7fqs6rd5n1mn95l9swd40000gn/T/.arduinoIDE-unsaved2025614-18258-1ghf3yz.oq1e/msc_sd/msc_sd.ino:21:1: error: 'Adafruit_USBD_MSC' does not name a type; did you mean 'Adafruit_USBD_CDC'?
   21 | Adafruit_USBD_MSC usb_msc;
      | ^~~~~~~~~~~~~~~~~
      | Adafruit_USBD_CDC
/private/var/folders/fm/30cx7fqs6rd5n1mn95l9swd40000gn/T/.arduinoIDE-unsaved2025614-18258-1ghf3yz.oq1e/msc_sd/msc_sd.ino:23:1: error: 'Sd2Card' does not name a type; did you mean 'SdCard'?
   23 | Sd2Card card;
      | ^~~~~~~
      | SdCard
/private/var/folders/fm/30cx7fqs6rd5n1mn95l9swd40000gn/T/.arduinoIDE-unsaved2025614-18258-1ghf3yz.oq1e/msc_sd/msc_sd.ino:24:1: error: 'SdVolume' does not name a type; did you mean 'FsVolume'?
   24 | SdVolume volume;
      | ^~~~~~~~
      | FsVolume
/private/var/folders/fm/30cx7fqs6rd5n1mn95l9swd40000gn/T/.arduinoIDE-unsaved2025614-18258-1ghf3yz.oq1e/msc_sd/msc_sd.ino: In function 'void setup()':
/private/var/folders/fm/30cx7fqs6rd5n1mn95l9swd40000gn/T/.arduinoIDE-unsaved2025614-18258-1ghf3yz.oq1e/msc_sd/msc_sd.ino:32:3: error: 'usb_msc' was not declared in this scope
   32 |   usb_msc.setID("Adafruit", "SD Card", "1.0");
      |   ^~~~~~~
/private/var/folders/fm/30cx7fqs6rd5n1mn95l9swd40000gn/T/.arduinoIDE-unsaved2025614-18258-1ghf3yz.oq1e/msc_sd/msc_sd.ino:43:7: error: 'TinyUSBDevice' was not declared in this scope; did you mean 'TinyUSB_Device_Task'?
   43 |   if (TinyUSBDevice.mounted()) {
      |       ^~~~~~~~~~~~~
      |       TinyUSB_Device_Task
/private/var/folders/fm/30cx7fqs6rd5n1mn95l9swd40000gn/T/.arduinoIDE-unsaved2025614-18258-1ghf3yz.oq1e/msc_sd/msc_sd.ino:53:9: error: 'card' was not declared in this scope
   53 |   if ( !card.init(SPI_HALF_SPEED, chipSelect) ) {
      |         ^~~~
/private/var/folders/fm/30cx7fqs6rd5n1mn95l9swd40000gn/T/.arduinoIDE-unsaved2025614-18258-1ghf3yz.oq1e/msc_sd/msc_sd.ino:62:8: error: 'volume' was not declared in this scope; did you mean 'FsVolume'?
   62 |   if (!volume.init(card)) {
      |        ^~~~~~
      |        FsVolume
/private/var/folders/fm/30cx7fqs6rd5n1mn95l9swd40000gn/T/.arduinoIDE-unsaved2025614-18258-1ghf3yz.oq1e/msc_sd/msc_sd.ino:62:20: error: 'card' was not declared in this scope
   62 |   if (!volume.init(card)) {
      |                    ^~~~
/private/var/folders/fm/30cx7fqs6rd5n1mn95l9swd40000gn/T/.arduinoIDE-unsaved2025614-18258-1ghf3yz.oq1e/msc_sd/msc_sd.ino:67:26: error: 'volume' was not declared in this scope; did you mean 'FsVolume'?
   67 |   uint32_t block_count = volume.blocksPerCluster()*volume.clusterCount();
      |                          ^~~~~~
      |                          FsVolume
/private/var/folders/fm/30cx7fqs6rd5n1mn95l9swd40000gn/T/.arduinoIDE-unsaved2025614-18258-1ghf3yz.oq1e/msc_sd/msc_sd.ino: In function 'int32_t msc_read_cb(uint32_t, void*, uint32_t)':
/private/var/folders/fm/30cx7fqs6rd5n1mn95l9swd40000gn/T/.arduinoIDE-unsaved2025614-18258-1ghf3yz.oq1e/msc_sd/msc_sd.ino:88:10: error: 'card' was not declared in this scope
   88 |   return card.readBlock(lba, (uint8_t*) buffer) ? 512 : -1;
      |          ^~~~
/private/var/folders/fm/30cx7fqs6rd5n1mn95l9swd40000gn/T/.arduinoIDE-unsaved2025614-18258-1ghf3yz.oq1e/msc_sd/msc_sd.ino: In function 'int32_t msc_write_cb(uint32_t, uint8_t*, uint32_t)':
/private/var/folders/fm/30cx7fqs6rd5n1mn95l9swd40000gn/T/.arduinoIDE-unsaved2025614-18258-1ghf3yz.oq1e/msc_sd/msc_sd.ino:96:10: error: 'card' was not declared in this scope
   96 |   return card.writeBlock(lba, buffer) ? 512 : -1;
      |          ^~~~
Multiple libraries were found for "SD.h"
  Used: /Users/leah/Library/Arduino15/packages/rp2040/hardware/rp2040/4.6.0/libraries/SD
  Not used: /Users/leah/Library/Arduino15/libraries/SD
  Not used: /Users/leah/Documents/Arduino/libraries/SD
Multiple libraries were found for "SdFat.h"
  Used: /Users/leah/Library/Arduino15/packages/rp2040/hardware/rp2040/4.6.0/libraries/SdFat
  Not used: /Users/leah/Documents/Arduino/libraries/SDFat_adafruit
Multiple libraries were found for "Adafruit_TinyUSB.h"
  Used: /Users/leah/Documents/Arduino/libraries/Adafruit_TinyUSB_Arduino
  Not used: /Users/leah/Library/Arduino15/packages/rp2040/hardware/rp2040/4.6.0/libraries/Adafruit_TinyUSB_Arduino
exit status 1

Compilation error: 'Adafruit_USBD_MSC' does not name a type; did you mean 'Adafruit_USBD_CDC'?
```

zachtos, 2026-01-20:
meh this doesn't work on ESP32S3 platformIO platform = espressif32 @ 6.12.0
board = esp32-s3-devkitc-1
C:/Users/zsdic/OneDrive/Documents/PlatformIO/Projects/ESP32_Ubox/src/main.ino:55:1: error: 'Sd2Card' does not name a type; did you mean 'SdCard'?
 Sd2Card card;
 ^~~~~~~
 SdCard
C:/Users/zsdic/OneDrive/Documents/PlatformIO/Projects/ESP32_Ubox/src/main.ino:56:1: error: 'SdVolume' does not name a type; did you mean 'FsVolume'?
 SdVolume volume;